darthcloud/BlueRetro #834: Mouse Support for the Atari ST

Sowden, 2023-11-18:
### BlueRetro firmware version

v1.9.hw1 parallel_2p

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

GrechTech

### System used

Atari 2600

### Bluetooth controller brand & name

Retro Rosetta

### What is problem? (only list ONE problem per report)

This is actually for the Atari STE, not the 2600, but the connection is the same port.

A bluetooth mouse connects to my device, but it does not move anything at all.  I can set my PS4 controller left thumbstick to up down, left right, and it does move the mouse.  But only one inch per moving the thumbstick in that direction.  I can only get it to work on that controller, but it won't move at all on at wireless mouse (that I know it connects properly to my Retro Rosetta because it indicates that its connected to Device 1, and my PS4 controller would then connect to Device 2).

### What did you expect to happen?

I expect it to move.  What I think sets this mouse apart from any other one is that this is connected to a DB9M connector, and the mouse ball moves the joystick (up down left right), because if you plug a joystick into the same port it does the same thing.  You can read more about the joystick/mouse port here:

https://info-coach.fr/atari/hardware/interfaces.php#The_Keyboard_Interface

What I think is needed is not only some assistance in making the mouse move smoothly.  But I also think that a preset may need to be created.  Because the up down left right is inverted, and thats because thats how mouses with balls inside work.  So a preset would  be gratefully appreciated.

### Attach files like logs or Bluetooth traces here

If I can provide a log, let me know how and I will gladly post it.

Sowden, 2023-12-11:
Alright, so I've updated to the latest firmware. I've done some more experimenting, and I have had some success, but not fully there yet.

First off I got the mouse to work with my PS4 controller.  What I had to do first was set the right stick to Output 1, because the mouse is always plugged into the first port (Atari port 0).  Then I had to set the directions in a complete reverse

`Right stick left -> D-pad up`
`Right stick right -> D-pad down`
`Right stick down -> D-pad right`
`Right stick up -> D-pad left`

Thats cause the mouse reads the directions as a normal joystick input, and the directions are reverse because the wheels on the inside are upside down.  That at least got it to move in any direction.  I then had to set Turbo for all directions to "1/2 Frames" to make it move at a slow pace.  With all of that, the PS4 right stick could move the mouse on screen.  And the buttons can be mapped to R2 and L2.  So then what about a real mouse?

Well thats still not working all that great.  With all of those settings as is, I can only get the mouse to move up and left directions.  Down and right doesn't move at all.  Changing the Max and the Deadzone doesn't seem to make any changes in that.  What does change though is the Threshold, and only if you change it to 0% (as far as I can tell at least).  Changing that does make all the directions move, but it moves in a very floaty way, and doesn't stop its directions with the move of the mouse.  Changing the Turbo only seems to make it move more slowly at this point.  It doesn't make it move any faster, which is still a bit of a problem.  But the mouse clicks do work, so at least there is that.

So hey @darthcloud, does any of this make sense to you?  I feel like I am starting to run out of options on how to experiment with this mouse.  Do you have any suggestions or guidance?  Thanks man.

darthcloud, 2023-12-11:
There is no mouse support for 2600.

What you are doing is just mapping the mouse inputs as atari joystick buttons.

And that unlikely to work well.

Sowden, 2023-12-11:
> What you are doing is just mapping the mouse inputs as atari joystick buttons.
> 
> And that unlikely to work well.

Yes thats because the Atari mouse is plugged into a DB9M connector.  Could you elaborate a bit on why it won't work very well?

darthcloud, 2023-12-11:
Its the same connector and pins but a different protocol, Joystick are simple buttons in parallel while a mouse use quadrature signaling.

BlueRetro doesn't support quadrature signaling. So it's not possible.

Sowden, 2023-12-11:
Ok, thanks for your reply.  I guess my last questions would be with keeping the quadrature signaling in mind, would you have any idea why changing the Threshold to 0% would make such a drastic change for floaty mouse movements?

closed 2024-04-15 by darthcloud