darthcloud/BlueRetro #843: No player 2 connection

thc1998, 2023-12-01:
### BlueRetro firmware version

1.9

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

Me

### System used

SEGA MegaDrive Genesis

### Bluetooth controller brand & name

Ps5 

### What is problem? (only list ONE problem per report)

When connecting controllers it jumps straight from player 1 to player 3 on hw2 ive tried both universal and genesis firmware on hw1 everything works except reset obviously 😅

### What did you expect to happen?

Connect to player 2

darthcloud, 2024-04-15:
You likely need to use HW1 firmware.

Sound like you don't have the port detecting wiring that HW2 firmware require.

darthcloud, 2024-10-06:
Feel free to reopen if you got more info

closed 2024-10-06 by darthcloud