darthcloud/BlueRetro #847: Blueretro N64 Memory Pak not working properly for some games.

Xenphor, 2023-12-04:
### BlueRetro firmware version

1.9

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

8bitmods

### System used

Nintendo 64

### Bluetooth controller brand & name

Unknown

### What is problem? (only list ONE problem per report)

Top Gear Rally does not consistently update new saves. Sometimes it will stick, other times will not. Dumping and restoring the save does not help. 

Robotron 64 consistently does not update a new save.

### What did you expect to happen?

I expect to create a new save over a previously saved game and for the Blueretro adapter to update and retain the new save for loading.

### Attach files like logs or Bluetooth traces here

Here is a [video ](https://youtube.com/shorts/A08iDybPDIU?feature=share) of it happening in Robotron 64.

Xenphor, 2023-12-06:
> I tried Robotron and it always work for me.
> 
> Do you have an unpatched N64 Digital HDMI mod installed?

No, I have an unmodded N64.

darthcloud, 2023-12-06:
Saw your video only now, I did try overwriting the save between levels, and they do update fine for me.

You might want to try latest beta:
https://github.com/darthcloud/BlueRetro/actions/runs/7081676820

Xenphor, 2023-12-08:
I've tried the new beta and the issue still happens in Top Gear Rally. Robotron 64 seemed to still have the problem as well, but I think it seems to have weird behavior with rumble pack and memory pak switching. It might have thought the memory pak was inserted when it actually wasn't. I was able to reproduce the problem but not consistently. 

Top Gear Rally still won't save after completing the 4th season, 2nd year. I dumped the memory pak right after I saved at the beginning of the 5th season and restoring that image put me back at the beginning of the 4th season. I don't know if this is how the game behaved with an original memory pak. It says it saves after completing each race.

My Quake 2 save has been able to consistently load, even after restoring the memory pak image.

rpierucci, 2024-01-30:
I had the same issue. The only way I found to resolve it was to go into the BlueRetro configuration and I had my Accessories set to Both. Once I set it to just Memory, games started saving and I didn't have a problem. Hope this helps in some way.

Xenphor, 2024-01-31:
I did have it set to both, so I might try memory only in the future. Although I've kind of stopped playing Top Gear Rally at the moment.

RogerEngineer99, 2025-03-12:
> I tried Robotron and it always work for me.
> 
> Do you have an unpatched N64 Digital HDMI mod installed?

I'm experiencing somewhat similar issues. I bought the blueretro adapters recently.  And for some games like Cruisin the USA it works fine. But in games like Bug's Life and Tony Hawk Pro Skater it will refuse to work. I can't seem to figure out what I have to do make it work properly. I have also flashed latest firmware and configured it advanced configuration to have both accessories rumble and data pak. My Nintendo 64 is the Pickachu US edition and has the Pixel FX board installed for HDMI output. I had to reply to your comment I don't know what you about unpatched N64 Digital HDMI mod... And is that an issue for these adapters?

marcgd, 2025-04-23:
I’m having the same issue on latest release version. When I set Accessories to Both, neither saving nor rumble works, and the Memory Pak isn’t detected. But when set to Memory, saving works fine, and Rumble gives me rumble. Seems like both settings don’t play nice together.

darthcloud, 2025-04-23:
@marcgd Both is for Dreamcast, use either memory card or rumble like a real n64, you can't plug both.