darthcloud/BlueRetro #849: ZL and ZR not working as Trigger when used in Dual-Mode

ParadoxLULP, 2023-12-08:
### BlueRetro firmware version

1.9 HW1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

RetroScaler NGC4

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Nintendo Switch Joy-Con

### What is problem? (only list ONE problem per report)

Hello, I'm completely new to this community and tried to find a solution in the different issues of this project.
I just received my BlueRetro NGC 4 from RetroScaler, and after updating to the newest version mentioned here, the problem of the disconnecting second Joy-Con stopped completely. Everything stays connected. 
But whenever I try to use the Dual-Mode, everything works except the mapping for the GC L/R.
This results in the problem, that I can't drift in MK: DD, one of the two games i have.
What I already tried is remapping the Gamecube Digital and Analog L and R to different buttons of a Joy-Con, but it never shows in the game, that the trigger is pressed. 
If I load the gamecube merge analog and digital preset, and only play with one Joy-Con, everything works. Using a PS5 controller also doesn't show any problems. 

### What did you expect to happen?

The LT (ZL) and RT (ZT) should work in Dual-Mode the same as in Single-Mode.
Are there any setting to change to work around this problem? 
I also don't have experience in modding or similar, so I use a stock DOL-001 (EUR) with a retrobit prism hdmi adapter. 
If any more information are needed, I try to answer as fast as possible. 
Thanks in advance

ParadoxLULP, 2023-12-13:
After some additional trying it occurs that as soon there is a second controller connected, the R and L of the gamecube doesn't works anymore. 
Tried it with the following set ups:
- Joycon 1 and 2 as output 1 (one time directly with the preset configuration for joy cons and one time with the merger preset and then reconfigure device 2 to also output 1) plus a Ps5 controller as output 2
- Ps5 as output 1, joycon 1 as output 2 and joycon 2 as output 3.
Everytime i connect more than one controller, the L/R mapping stops working.
I can put the Z mapping on every button where I put the L/R button on before and the Z works. Only the L/R stops. 
Anyone got an idea what I can try now?

psicotron3, 2023-12-22:
I have the same issue on a wii (gamecube ports), using 2 xbox controllers, and found out that:

1) I need to map digital (RB) and analog triggers (RT) to a button (RB). If I map only digital triggers (RB) to digital triggers (RB) I won't register the pressed button.
2) Mapping digital and analog triggers to a button, it will register the digital and analog trigger press as expected. But only if there’s only one controller connected.
3) If I connect a second controller, both controllers will only register the analog trigger pressing. Digital trigger button press is ignored.

Maybe, it's related to the fix for issue #651 that added a "Delay GC digital trigger by 2 frames vs analog portion".

ParadoxLULP, 2023-12-22:
Do I have to add this fix myself at my adapter? I'm new to all this stuff and updating to the newest firmware was already a struggle for me, although I got it to work.

psicotron3, 2023-12-22:
Sorry. That's not a fix for the issue. I just pointed out that the previous change may be related to the issue.

By the way, is your adapter, like mine, like the one showed here: #565

That's just like mine and this adapter and Retroscaler haven't follow this project hardware documentation for the leds. God knows what else they did.

If so, the issue isn't with the code but the way the adapter works.

They have their fork on githhub fork: https://github.com/RetroScaler/NGC-BlueRetro.

Maybe you can try updating the firmware with the last version on their release page. 

I'll try that later.

ParadoxLULP, 2023-12-22:
Yes, i think I have the same. Atm I'm unable to test stuff due to work, but please keep me updated if you are able to test it.

psicotron3, 2023-12-23:
> Maybe, it's related to the fix for issue #651 that added a "Delay GC digital trigger by 2 frames vs analog portion".

I can confirm that reverting the changes in the mentioned issue fix (#651) fixed the issue where, with more than 1 controller connected digital shoulder buttons stops responding.

While testing I also found out that, if the digital buttons press are registered if I press the same button on all controller.

Maybe there's a better logic for this implementation, but it's way beyond what I can do and it's yet to be confirmed that the issue happens with all adapters or only the faulty retroscaler one.

For @ParadoxLULP - installing a older version of the firmware will be a temporary solution for the issue. Maybe 1.8.2 release from darthcloud repo. Although I haven't tested any of the retroscaler versions. 

Anyhow, I don't think using two switch controllers is a good option, you'll find a lot of other issues.

darthcloud, 2023-12-23:
Just don't try to use 2 joycon as one controller, its really the worst setup possible for BlueRetro as their is an issue with using 2 switch controller that I haven't figured yet. On top of that the joycon preset doesn't merge the analog and digital trigger. So you will have to do it manually.

Switch controller are very a poor choice for gamecube in general has they dont have analog triggers. Its tempting to use the various GC styled ones but they do not provide analog triggers.

Ultimately you can do whatever you want but mapping the GC triggers to a digital button proven to more challenging than I thought . For example Metroid Prime expect the analog part to be pressed a couple frame before the digital end. Otherwise it ignore the digital end. If you press a OG ctrl trigger real quick you can reproduce that behavior btw. The fix you linked to was for that.

So if you map only the digital part I can see that some games will ignore it since the analog part is not pressed.

I will look more into it but meanwhile try use a controller with analog trigger while using the gc merge trigger preset.

ParadoxLULP, 2023-12-23:
Thanks for the reply, and I really hope that some day you will find a fix for that.

Then for the moment I will not use joy cons, and I try to find time in the next days to test it again. The last time I tried to use multiple controller was 1 PS5 and 2 joy cons separated with all 3 outputs set up as the trigger merge preset. 

I don't know if I did anything else wrong, because it was late at this point,  but even then the trigger didn't work as trigger, only the Z of the gamecube worked regardless where I mapped it manually. 

I'm moving in some days, so I wish you all a very good Christmas and a happy new year now before I forget it later :)

ValadAmoleo, 2023-12-23:
Does anybody have 1.8.2 available? Itch unfortunately only goes back to 1.8.3 and would be nice to use my Nyxi Wizard with Mario Kart Double Dash. I wish there was a Bluetooth GameCube-shaped pad available with analogue triggers but unfortunately there isn't at the moment, so dual joycon mode with the Nyxi Wizard is the only wireless option.

Selmaks, 2024-02-06:
> I have the same issue on a wii (gamecube ports), using 2 xbox controllers, and found out that:
>
>     1. I need to map digital (RB) and analog triggers (RT) to a button (RB). If I map only digital [...]

I am facing a similar issue using ps5 controllers, I got my core and GG cables from https://vi.aliexpress.com/item/1005004661877197.html . If I use 1 controller with L2 analog at 50% L2 digital at 95% it works. If I use two controllers the digital doesn't seem to work unless both controllers are pressing L2 digital at the same time it's like that button is somehow linked on both controllers. If I use [1.8.2](https://github.com/darthcloud/BlueRetro/releases/download/v1.8.2/v1.8.2_hw1.zip) it seems to work without that issue.

darthcloud, 2024-04-15:
This beta should fix it:
https://github.com/darthcloud/BlueRetro/actions/runs/8680779369

ParadoxLULP, 2024-04-17:
Thank you so much!!! Today I had time to test the beta. I used my Nxyi Wizard gamecube/joycon controller. With the presets for the upright joycons and some tweaking in the advanced setting page, i can use it (in my limited experience) lag free as a whole Gamecube Controller. Thank you so much for your work. Also the problem of the disappearing Triggers while connecting more than 1 controller is not there while using the gamecube trigger merge preset anymore.

darthcloud, 2024-04-24:
Fixed in v24.04

closed 2024-04-24 by darthcloud