darthcloud/BlueRetro #856: Update not finishing properly

myeuropeanPAL, 2023-12-10:
### BlueRetro firmware version

v1.9.1_hw1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

n/a

### System used

Nintendo GameCube

### What is problem? (only list ONE problem per report)

The update never seems to finish properly via https://blueretro.io/ota.html

It says 100% but the live out stops here and never disconnects:
[image]

Requesting Bluetooth Device...
Connecting to GATT Server...
Getting BlueRetro Service...
app_ver_is_hw2: false app_name_is_hw2: false
Init Cfg DOM...
new_fw_is_hw2: false

However after disconnecting the from usb, the device seem to have the latest firmware.

[image]

### What did you expect to happen?

Disconnecting

darthcloud, 2023-12-10:
This varies from pc to pc.

My usual advice is to try another computer or try to do it in another location if you are in a crowded area BT wise.

But since you got and USB port just update it via USB!
https://github.com/darthcloud/BlueRetro/wiki/Bluetooth-HCI-trace-with-Win10

myeuropeanPAL, 2023-12-11:
Hello, thanks for the quick reply, the same environment and everything works just fine with a PS2 styled blueretro adapter, that is why I thought it is worth reporting this, but I will sick to the usb update for the future. Thanks!

WolfRamiO74, 2024-01-02:
same here with retroscaler Blueretro PS2 mini with USB, if i connect it to the PC via USB the device is not detected.

darthcloud, 2025-01-24:
This got fixed this fall, update over usb to [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)

After that OTA update should work.

closed 2025-01-24 by darthcloud