darthcloud/BlueRetro #857: Support for SN30 Pro+ and (Bitdo Arcade Stick broken

GeneralMorden, 2023-12-11:
### BlueRetro firmware version

1.9.1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Humble Bazooka and Laserbear

### System used

SNK NeoGeo AES

### Bluetooth controller brand & name

SN30 Pro+ and Arcade Stick 8Bitdo

### What is problem? (only list ONE problem per report)

In X-Input SN30 Pro+ and Arcade Stick are not support since 1.9.1. Mapping is broken.

### What did you expect to happen?

Please fix this. Is the a place i can download 1.9 firmware?

darthcloud, 2023-12-11:
I re-added v1.9 on https://darthcloud.itch.io/blueretro for now.

darthcloud, 2024-01-02:
Can you try this beta and see if the mapping is right:
[v1.9.2-beta-3-g4135621_hw1.zip](https://github.com/darthcloud/BlueRetro/files/13812810/v1.9.2-beta-3-g4135621_hw1.zip)
[v1.9.2-beta-3-g4135621_hw2.zip](https://github.com/darthcloud/BlueRetro/files/13812811/v1.9.2-beta-3-g4135621_hw2.zip)

GeneralMorden, 2024-01-07:
> [Verfolgungsbild][https://github.com/notifications/beacon/AO7XYPK5FYY2CGAQNH57VDLYMREH7A5CNFSM6AAAAABAQLU2JKWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTPXCOD6.gif]

Yes i will try.

GeneralMorden, 2024-01-07:
Tested the beta per 8Bitdo Arcade Stick on NeoBT (MVS) and SN30 Pro on Laserbears Game Cube device. Now, everything seems to Work correctly 👍

darthcloud, 2024-02-19:
[Fixed in v1.9.2](https://github.com/darthcloud/BlueRetro/releases/tag/v1.9.2)

closed 2024-02-19 by darthcloud