darthcloud/BlueRetro #869: Cannot connect n64 adapter to blueretro.io

BloatedPenguin, 2023-12-23:
### BlueRetro firmware version

1.8.4

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler

### System used

Nintendo 64

### What is problem? (only list ONE problem per report)

Web interface sees the adapter, but will not connect. "GATT operation failed for unknown reason."

### What did you expect to happen?

To connect the adapter to blueretro.io and adjust button mapping.

### Attach files like logs or Bluetooth traces here

[image]

BloatedPenguin, 2023-12-23:
I've been trying for hours to get this thing to connect. Wiped the adapter of any controllers, tried different computers, browsers, rolling back Chrome to ver. 104, connecting to a different network, but I'm still getting the above message. Anything I'm missing?

cpawliuk, 2023-12-28:
I would just update the firmware manually at this point since you're on 1.8.4: https://github.com/darthcloud/BlueRetro/wiki/Flashing-firmware-Windows-10 Then try again?

I wonder if it is even possible to edit the config without using the site? Please confirm if this is possible and how.

BloatedPenguin, 2023-12-29:
Haven't tried the suggestion yet, but now I'm getting this message.
[image]

cpawliuk, 2023-12-29:
I would still do the manual firmware update. That error I believe has to do with failure to connect with the web-bluetooth service. Try in another tab maybe? I've had problems since day one connecting and it was a hit or miss. Firmware updates were also a pain and flashing to another because it would drop before 100%. Just do the manual way and try again?

For the profile mapping and stuff I'm guessing if there isn't an offline way provided one would need to make a custom firmware with those settings? Not sure what is on the server side for the site as I haven't looked but I guess one could rip the site and if the profiles are handled with JSON it shouldn't be much to edit and make that work.

BloatedPenguin, 2023-12-29:
I'm not quite sure how to do a manual firmware update when it won't connect to a computer via bluetooth. Do I need to break it out of its casing to plug it in directly to the computer?
Also, Doing a bit more research, since it is technically a "Retro Scaler" adapter their website says to not update it with the Darth Cloud firmware but to use theirs, so this could be why I'm having issues..

cpawliuk, 2023-12-29:
I'm not 100% sure with RetroScaler but they have instructions for the N64 one here to do it manually:
https://github.com/RetroScaler/N64-Blueretro

FairyKid64, 2024-02-20:
I'm also having this problem.  I have been able to connect this RetroScaler N64 adapter before, but then I updated to I think 1.8.2 if I remember correctly, and now I can't even connect to Chrome to update again.  These are the log messages it gives me:

Requesting Bluetooth Device...
Connecting to GATT Server...
Getting BlueRetro Service...
app_ver_is_hw2: false app_name_is_hw2: false
Init Cfg DOM...
> Bluetooth Device disconnected

closed 2024-04-15 by darthcloud