darthcloud/BlueRetro #875: Controllers not Working in Shmups

ghiblinova, 2023-12-31:
### BlueRetro firmware version

1.9.1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

HumbleBazooka

### System used

SEGA Saturn

### Bluetooth controller brand & name

8BitDo DIY Saturn kit

### What is problem? (only list ONE problem per report)

Many shmups do not work, neither in D-pad mode nor Analog mode. Radiant Silvergun, Batsugun, Galactic Attack, Hyper Duel, In The Hunt, Sengoku Blade, Strikers 1945 II, I'm sure there's more and I'd be happy to test all shmups for the project but I think this is good for now.

### What did you expect to happen?

I expected to play the game. For some reason as soon as the games boot up, all inputs stop working.

### Attach files like logs or Bluetooth traces here

I'm sorry, I don't know what this means.

DoomyDoomer, 2023-12-31:
I wasn't able to replicate the issue in Radiant Silvergun using my 8BitDo DIY kit. I don't have the other games you listed but I'll track them down for testing. Maybe you have the DIY kit in a weird mode. Double check the DIY kit manual for and see if there are modes other than xinput/dpad. I can't recall if their DIY kits have other modes.

Also, I believe In The Hunt has a bug where if you have a controller plugged in while booting the game, it won't boot. This goes for all controllers, not just wireless adapters.

ghiblinova, 2023-12-31:
Thank you for getting back to me so quick! In the Hunt works with my wired
controllers, in fact all these games work for me with wired controllers.
Could there be a bug when booting games off a Satiator? And I'm booting the
controller by pressing start and X, then holding up or left with L+R to
toggle between D-pad and Analog mode.

darthcloud, 2024-01-02:
can you test with a stock saturn and a genuine game to see if it makes a difference?

darthcloud, 2024-04-15:
Feel free to reopen if you got extra info

closed 2024-04-15 by darthcloud