darthcloud/BlueRetro #876: Dual Shock 3 Sixaxis Controller Does Not Pair

grub3r-codes, 2024-01-02:
### BlueRetro firmware version

1.9.1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

retroscaler

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Sony Dualshock 3 Sixaxis

### What is problem? (only list ONE problem per report)

Controller doesn't pair

### What did you expect to happen?

Controller should pair

### Attach files like logs or Bluetooth traces here

```
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_CONN_REQUEST
# bt_hci_cmd_accept_conn_req
# Page dev: 0 type: 0 bdaddr: 00:26:5C:50:4A:19
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_ROLE_CHANGE
# BT_HCI_EVT_CONN_COMPLETE
# dev: 0 acl_handle: 0x0080
# bt_hci_cmd_le_set_adv_disable
# bt_hci_cmd_remote_name_request
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_STATUS
# BT_L2CAP_CONN_REQ
# BT_L2CAP_CONF_REQ
# BT_HCI_EVT_INQUIRY_COMPLETE
# bt_hci_cmd_exit_periodic_inquiry
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_le_set_scan_param_passive
# bt_hci_cmd_le_set_scan_enable
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_REMOTE_NAME_REQ_COMPLETE:
# dev: 0 error: 0x22
# bt_hci_cmd_remote_name_request
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_REMOTE_NAME_REQ_COMPLETE:
# dev: 0 error: 0x02
# bt_hci_cmd_remote_name_request
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_REMOTE_NAME_REQ_COMPLETE:
# dev: 0 error: 0x02
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_le_set_scan_param_active
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_periodic_inquiry
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_DISCONN_COMPLETE
# DISCONN for non existing device handle: 0080
# BT_HCI_EVT_LE_ADVERTISING_REPORT
```

grub3r-codes, 2024-01-02:
Okay what's interesting is that after I hardware reset the controller, the name is properly detected but then BR stops logging:
```
# BT_HCI_EVT_CONN_REQUEST
# bt_hci_cmd_accept_conn_req
# Page dev: 0 type: 0 bdaddr: 00:26:5C:50:4A:19
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_ROLE_CHANGE
# BT_HCI_EVT_CONN_COMPLETE
# dev: 0 acl_handle: 0x0081
# bt_hci_cmd_le_set_adv_disable
# bt_hci_cmd_remote_name_request
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_STATUS
# BT_L2CAP_CONN_REQ
# BT_L2CAP_CONF_REQ
# BT_HCI_EVT_INQUIRY_COMPLETE
# bt_hci_cmd_exit_periodic_inquiry
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_le_set_scan_param_passive
# bt_hci_cmd_le_set_scan_enable
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_REMOTE_NAME_REQ_COMPLETE:
# bt_type_update: dev: 0 type: 1 subtype: 0
# bt_hci_cmd_exit_periodic_inquiry
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_le_set_scan_param_passive
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_switch_role
# dev: 0 type: 1:0 PLAYSTATION(R)3 Controller
# BT_HCI_EVT_CMD_COMPLETE
# opcode: 0x0404 error: 0x0C retry: 0
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_ROLE_CHANGE
```

grub3r-codes, 2024-01-02:
But this doesn't seem to be repeatable, I get the first set of logs 99% of the time.

Nizdazen, 2024-01-09:
Dealing with the exact same issue, I bought 2 of retroscaler's dongles for the ps2 and I can't pair either with my sixaxis dualshock 3-was trying to get the aesthetics of the original ps2. On the blueretro page it says Connected to: BlueRetro_PS2_BEBA followed by the Mac address correct? I put that into the sixaxis tool and successfully changed the master for the sixaxis controller but every time I then plug in the dongle to the ps2 and press and hold the pair buttons nothing pairs. I keep trying with either dongle, updating the firmware, resettting to factory default, nothing seems to work. Same model number controller cechzc2e. I've paired controllers with retroblue on the sega saturn, and N64 but I'm having no luck here on this one.

Nizdazen, 2024-01-09:
Tried downgrading to 1.8.3 & updating mac address just in case. No luck.

Nizdazen, 2024-01-10:
Opened up the Sixaxis controller. IT was a counterfeit despite the seller on ebay selling it listed as oem and having a high rating.

Nizdazen, 2024-01-10:
ebay seller Casebeworth23 sold me the fake so I guess watch out for him, blueretro won't work with fake sixaxis

Nizdazen, 2024-01-10:
Oh and I googled the model number that was on the circuit board which was HX-BT214303-v1.4 and Shenzhen Oubang Industrial shows up as the company that made that counterfeit-some statement where they are declaring the item with the FCC. The quality is actually pretty good, ran pretty well on SPCtoolkit. but counterfeit sixaxis don't work with blueretro, so that's no good.

darthcloud, 2024-01-10:
Anything brand new PS3 everywhere is fake, and yes they even faked the packaging perfectly.

The bug is about CECHZC2EA which is a Euro/Australia model which I don't have any.
Assuming it's isn't a fake one, it's possible it got a different FW than the North American ones I got.

But I don't have one so I can't test myself.

Nizdazen, 2024-01-12:
I purchased 2 ps3 dualshocks from a source I trust this time and they both paired no promblem this time, tested them both on a ps2 slim with OPL running (through the ethernet on the back using a portable harddrive connected to a travel router). They run great, only thing is both light up as player 1 on the LED indicator lights even though one controls player one and the other controls player 2. On my blueretro for N64 with the Nintendo Online n64 Controllers the led player indicators light up correctly. Don't know if that's a hardware limitation of the ps3 controller.

darthcloud, 2025-04-16:
Feel free reopen once you got the trace.

closed 2025-04-16 by darthcloud