darthcloud/BlueRetro #879: Default mapping changes for system-specific builds

qwertymodo, 2024-01-03:
### BlueRetro firmware version

1.9.1

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Select a type

### Manufacturer

DIY

### System used

Nintendo SFC SNES

### Bluetooth controller brand & name

8bitdo SN30 "Classic"

### What is problem? (only list ONE problem per report)

The 8bitdo SN30/SFC30 "Classic" controllers do not have an analog stick, but by default they map their DPad to the left analog axes instead of mapping them as a DPad.  Because BlueRetro sees that a DPad exists in the HID descriptor, it expects directional inputs to come from that, so out of the box, the SN30/SFC30 are unable to send any directional inputs to BR.

### What did you expect to happen?

I expect the SN30/SFC30 DPad to work normally.

### Attach files like logs or Bluetooth traces here

This issue was discussed in Discord, and a few solutions were brought up.  First of all, you said you didn't want to mess around with the default mapping anymore, but seemed open to the idea of creating "factory default data" for the system-specific builds.  However you want to go about implementing it behind the scenes, here are the mapping changes I would suggest:

### For consoles that only have a DPad and no left analog (e.g. NES/SNES/Genesis/etc.):
In addition to mapping controller DPad to console DPad, also map controller left analog to console DPad

### For consoles with only a single set of shoulder buttons (e.g. SNES/Saturn/etc.):
Map all controller shoulder buttons and analog shoulder axes to the single console shoulder buttons (e.g. PS2 controller on SNES R1->R, R2->R)

qwertymodo, 2024-01-03:
Another possible (optional, much more subjective) mapping addition could be for consoles without a right analog stick, map all controller right analog inputs as face button turbo (e.g. SNES RU->X turbo, RD->B turbo, RR->A turbo, RL->Y turbo)

SinisterSticks, 2024-01-05:
Another problem comes with Virtual Boy and fight sticks.  When I try to map Xbox a b X and Y to right stick north south east west for hyper fighting it does not work. 

I'm mechanically did the right steps to get that working but the only things that seem to work are the d-pad and one of the two shoulder buttons.  The other the two shoulder buttons match the first instead of was different.

I think the sn30 is similar to a fight stick that it doesn't have analog directional inputs.

qwertymodo, 2024-01-05:
> I think the sn30 is similar to a fight stick that it doesn't have analog directional inputs.

Correct. The original SN30/SFC30 does not have any physical analog stick, it's modeled to look like an exact clone of the original SNES/SFC controller.  However, it includes an analog axis in its report descriptor, and defaults to mapping the physical DPad to the analog axis in the report descriptor.  Supposedly there's a button combination to switch it into digital mode where it maps to the DPad report, but I can't seem to get it working on mine.  Later models of the controller did add actual physical analog sticks.

darthcloud, 2024-02-19:
[Fixed in v1.9.2](https://github.com/darthcloud/BlueRetro/releases/tag/v1.9.2)

closed 2024-02-19 by darthcloud

SinisterSticks, 2024-03-22:
I'm having trouble downloading version 1.9.2 and I donated $2 for it on itch.io.  

I understand downloading might be different on a Macintosh in which case I need help downloading it onto my Virtual Boy Bluetooth adapter.

Any help?

Also will this fix my problem with the memory of the Virtual Boy Bluetooth adapter not remembering when I hand it off from Android or Mac to Xbox adaptive controller?