darthcloud/BlueRetro #889: BlueRetro stops working when a game on OPL runs PS2

TheReconJacob, 2024-01-10:
### BlueRetro firmware version

1.9.1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

BlueRetro from AliExpress

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

PS5 Controller

### What is problem? (only list ONE problem per report)

1. BlueRetro stops working and I can't use a controller when I run a game using OPL. (Originally reported issue on OPL Github https://github.com/ps2homebrew/Open-PS2-Loader/issues/1158)

### What did you expect to happen?

Controller to work when using OPL, given it works on normal disc games.

TheReconJacob, 2024-01-11:
I found a workaround to this for anyone that has the same problem. The cable I am using for my PS2 is the one that allows you to have two controllers connected to one blueretro device. For whatever reason, having the second cable slot in the player 1 port allowed it to work perfectly, and having the first cable slot in the player 2 port also worked for having 2 players. Basically just swap the cable round to the opposite of where you'd expect it to go. For some reason, this is the only way it works on OPL, and they don't work the normal way round. Whereas on disc, it doesn't matter which way round the cable is, the controllers will work either way.

dunny1g, 2024-05-04:
Same issue here on both v1.9.1 and v24.04 HW1, basically haven't been able to use my PS2 with a DualShock 3 since I got a BlueRetro months ago unfortunately.

> I found a workaround to this for anyone that has the same problem. The cable I am using for my PS2 is the one that allows you to have two controllers connected to one blueretro device. For whatever [...]

Couldn't get the workaround to work either, sounds like we have the same BlueRetro setup too.

darthcloud, 2025-05-13:
For me it work but I'm using their old official release from 2021:
sha1sum OPNPS2LD-v1.1.0.ELF
be594ff83a81486c20b3f7276bcef3041bea29f3 OPNPS2LD-v1.1.0.ELF

I can't get the latest betas of OPL to work on my FHDB setup, they just give me a black screen.

One thing you can try is disabling PADEMU feature at both the Global and game level in OPL settings.

closed 2025-07-13 by darthcloud

schmkl, 2025-08-04:
> One thing you can try is disabling PADEMU feature at both the Global and game level in OPL settings.

This worked for me, thank you!