darthcloud/BlueRetro #891: DC Keyboard ghost keystrokes in certain games.

polyv0x, 2024-01-10:
### BlueRetro firmware version

1.9.1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

Espressif

### System used

SEGA Dreamcast

### Bluetooth controller brand & name

Filco Majestouch Minila Air

### What is problem? (only list ONE problem per report)

PSO, ToTD seem to have a lot of "debounce" ghost keystrokes when typing fast. 

For example if I type asdf 5 times real quick, it will look like: "asdf asadf asdfa asdf asdfd"  (letters are randomly doubled, making typing correctly with higher speeds impossible)

Noticed that Quake 3 Arena has no issues whatsoever when typing fast in the chat. 

Same keyboard functions just fine on other hardware when typing fast.

### What did you expect to happen?

Accurate typing just like in Quake 3.

darthcloud, 2024-04-20:
Can you try this beta firmware:
https://github.com/darthcloud/BlueRetro/actions/runs/8762666480

I think it should fix it.

polyv0x, 2024-04-20:
Awesome work, this indeed fixed it :)

darthcloud, 2024-04-20:
Thanks for testing I will close once officially released

darthcloud, 2024-04-24:
Fixed in v24.04

closed 2024-04-24 by darthcloud