darthcloud/BlueRetro #892: Hori Wireless Controller detected as Lic Pro Controller but does not have same quirks

mjones-asasolutions, 2024-01-11:
### BlueRetro firmware version

1.9.1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

BlueRetro (AliExpress)

### System used

Nintendo GameCube

### Bluetooth controller brand & name

HORI Wireless HORIPAD Pro Controller for Nintendo Switch

### What is problem? (only list ONE problem per report)

Controller is recognized as "Lic Pro Controller" so primary action button mapping is rotated.

### What did you expect to happen?

Expected to be able to map buttons using web ui tool with actual face value for buttons.

### Attach files like logs or Bluetooth traces here

In Debug was able to track it down to:

_/BlueRetro/main/bluetooth/hidp/hidp.c_

    {"Lic Pro Controller", BT_SW, BT_SW_POWERA, BIT(BT_QUIRK_FACE_BTNS_ROTATE_RIGHT)},

Made adjustment to be:

     {"Lic Pro Controller", BT_SW, BT_SW_POWERA, 0},

Recompiled and flashed firmware to board.  This fixed the issue I was experiencing.

mikeyj110, 2024-01-12:
Folowing.

darthcloud, 2025-01-24:
Fixed in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)

No mapping are tied to the Lic Pro Controller. PowerA GC user are now required to change mappin or use the new face button mapping shortcut (pressing select or minus)

closed 2025-01-24 by darthcloud