darthcloud/BlueRetro #903: Analog switch not working on PS2

WolfRamiO74, 2024-02-02:
### BlueRetro firmware version

1.9.1 PS2

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Retroscaler Blueretro "mini" for PS2

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Microsoft  Xbox One S controller

### What is problem? (only list ONE problem per report)

Hello there, I have a "BlueRetro Mini" for PS2 from Retro scaler.

The thing is that I connect an Xbox One S controller to it and after remapping everything works fine, except for the "Analog" switch, which should activate and deactivate the analog on the PS2, this is mapped to the "Xbox" button by default and when you press it it seems to do nothing, I tried mapping it to the "view" button but it didn't work either...

I am using Firmware 1.9.1 for PS2

### What did you expect to happen?

Asigned "Analog" button should switch on and of "analog" on PS2

darthcloud, 2024-02-03:
Updating your Xbox controller to latest firmware will fix this:
Update firmware via the [Xbox accessories Win10 app](https://apps.microsoft.com/store/detail/xbox-accessories/9NBLGGH30XJ3).

WolfRamiO74, 2024-02-07:
Hi, my controller was updated before i post this, i check again and the app say "theres no updates"

Its the first Bluetooth XBOX Controller, those with Micro USB port, not USB C

CaduGimenes, 2024-02-10:
Hi, same problema here...

My XBOX Controller is the new version (Series S - USB C), i update the controller firmware, update the software version (from 1.8.3 to 1.9.1 hw1), and nothing made the analogue work.

WolfRamiO74, 2024-02-10:
> Updating your Xbox controller to latest firmware will fix this: Update firmware via the [Xbox accessories Win10 app](https://apps.microsoft.com/store/detail/xbox-accessories/9NBLGGH30XJ3).

acabo de encontrar que el switch funciona en los juegos nativos de PS2, incluso cuando son cargados via OPL, pero no funciona en el SNESticle Homebrew.

WolfRamiO74, 2024-02-10:
> Updating your Xbox controller to latest firmware will fix this: Update firmware via the [Xbox accessories Win10 app](https://apps.microsoft.com/store/detail/xbox-accessories/9NBLGGH30XJ3).

I just found that the switch works on native PS2 games, even when loaded via OPL, but it doesn't work on SNESticle Homebrew.