darthcloud/BlueRetro #910: ESP32-D0WD-V3 Not working

necrosaromx, 2024-02-14:
### BlueRetro firmware version

v1.9.1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

I'm building it by myself

### System used

Nintendo 64

### Bluetooth controller brand & name

WiiU Pro, XBox Series S/X, Joycon,8BitDo NeoGeo

### What is problem? (only list ONE problem per report)

I flashed a ESP32-D0WDQ (CP2102 Driver) successfully and Worked like a charm with FW v1.9.1_hw1 on a N64, but I'm facing problems with an ESP32-D0WD-V3 (CH340C Driver) variant while trying to pair controllers, even the webconfig cannot pair with it.

None of listed controllers pair, status led keeps flashing and logs show no details (attached file)

Are there special instructions to flash ESP32-D0WD-V3 variants?

### What did you expect to happen?

Controllers would pair with ESP32 ESP32-D0WD-V3

### Attach files like logs or Bluetooth traces here

[ESP32-D0WD-V3.txt](https://github.com/darthcloud/BlueRetro/files/14273924/ESP32-D0WD-V3.txt)

darthcloud, 2024-04-15:
ESP32-D0WD-V3 is actually the only officially supported variant. (But the older ones should work too)

You likely got an issue somewhere else in your installation.

closed 2024-04-20 by darthcloud

soufiane707, 2024-04-24:
it fails with dual sense as well ( ps5 controller )