darthcloud/BlueRetro #913: PS2 - Resident Evil 4 - Controller vibration triggers in menus and cut scenes without reason.

cpawliuk, 2024-02-18:
### BlueRetro firmware version

1.9.1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

BlueRetro Core

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Xbox One

### What is problem? (only list ONE problem per report)

When playing Resident Evil 4 any cut scene or menu will trigger rumble. This does not happen on either a PS1 or PS2 controller, only a bluetooth controller using BlueRetro.

There was another bug report in JAN 2023 but it was not addressed: https://github.com/darthcloud/BlueRetro/issues/576 

### What did you expect to happen?

Rumble should not occur when entering the menu or a cut scene. Disabling vibration in the menu does not work.

ButtonBasher92, 2024-02-22:
Can Confirm this is also the case with V1.9.2 HW1, Retroscaler BlueRetro Mini PS2, using a PS3 or 4 controller

Penq1989, 2024-03-17:
The problem is probably lack of analog vibration, in almost all racing games You can't play with vibration, because they always spin. Only brook adapters support this.

darthcloud, 2024-05-29:
Can you test this beta, This completely rework the rumble support
I think it might fix that issue aswell.

https://github.com/darthcloud/BlueRetro/actions/runs/9286878322

darthcloud, 2024-05-29:
Don't use HW2, base on ticket you got HW1 device

cpawliuk, 2024-05-29:
> Don't use HW2, base on ticket you got HW1 device

It was a typo, I meant HW1.

darthcloud, 2024-05-29:
ok I'll setup that game and see what wrong

ButtonBasher92, 2024-05-29:
> Can you test this beta, This completely rework the rumble support I think it might fix that issue aswell.
> 
> https://github.com/darthcloud/BlueRetro/actions/runs/9286878322

I also get the same with Res4, otherwise dualshock with variable speed is working great so far in other games and tested with padtest.elf on the ps2.

thank you for this. is a much-appreciated feature.

darthcloud, 2024-05-30:
This one should fix RE4:
https://github.com/darthcloud/BlueRetro/actions/runs/9304053152

cpawliuk, 2024-05-30:
> This one should fix RE4: https://github.com/darthcloud/BlueRetro/actions/runs/9304053152

Works without issue now! Thanks for all your great work!

closed 2024-05-30 by cpawliuk