darthcloud/BlueRetro #930: Gamecube digital ‘click’ L and R Triggers

haydnkilyan, 2024-03-02:
### BlueRetro firmware version

v1.9.2

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

Internal install modkit

### Manufacturer

Laserbear

### System used

Nintendo GameCube

### Bluetooth controller brand & name

8Bitdo Pro 2

### What is problem? (only list ONE problem per report)

I set the L2 / R2 trigger threshold to act as the digital L/R trigger presses, when the threshold of 95% is reached. I have some controller test software to check and it works perfectly when one controller is connected. When I connect a second controller, the digital ‘click’ no longer registers on EITHER controller……. This was not a problem on the v1.8.3 firmware…. Though rumble does not seem to work on 8bitdo Pro 2 using this firmware…. Was going to roll back to v1.8.2 - but it is no longer available 

### What did you expect to happen?

The digital clicks to register at 95%

haydnkilyan, 2024-03-02:
What's sad is that I think v1.8.2 work perfectly for my set up - and now I cannot get it back!

haydnkilyan, 2024-03-03:
Just remembered, on v1.8.3 you have to set the controllers to have 'rumble' manually - so all works fine on 1.8.3...... but still issue on 1.9.X

darthcloud, 2024-04-09:
What is the 2nd controller brand you are issue? Both are 8bitdo Pro 2? Are they connected in Xinput mode?

All the old FW are in the release section:
https://github.com/darthcloud/BlueRetro/releases/tag/v1.8.2

haydnkilyan, 2024-04-10:
Hello! Thank you for getting back to me. Yes both controllers are 8Bitdo Pro 2s set to X input

darthcloud, 2024-04-11:
Does applying the GC Merge trigger preset help?

See animated gif in wiki
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#33---example-preset-load

darthcloud, 2024-04-14:
This beta should fix it there was indeed a problem with trigger in multiplayer:
https://github.com/darthcloud/BlueRetro/actions/runs/8680779369

darthcloud, 2024-04-24:
Fixed in v24.04

closed 2024-04-24 by darthcloud