darthcloud/BlueRetro #933: 8bitdo NeoGeo pad shoulder buttons not detected

GrumpyGopher, 2024-03-09:
### BlueRetro firmware version

1.9.2

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

Self

### System used

Sony PlayStation 1

### Bluetooth controller brand & name

8bitdo NeoGeo Pad

### What is problem? (only list ONE problem per report)

Left and Right shoulder buttons are being ignored with this controller.  This was working in firmware version 1.8.4 but starting in 1.9.0 and is still happening in 1.9.2

### What did you expect to happen?

When it was working in version 1.8.4 the left and right shoulder buttons would trigger the same as L1 and R1 on a 8bitdo SN30 pro+ controller.

darthcloud, 2024-04-09:
I goal was to have them on the main trigger L2/R2 but I didn't test it properly.

That should be fixed in this build:
https://github.com/darthcloud/BlueRetro/actions/runs/8609240832

darthcloud, 2024-04-24:
Fixed in v24.04