darthcloud/BlueRetro #952: Blueretro won't respond with v1.9.2_hw1_universal

asweintraub, 2024-04-08:
### BlueRetro firmware version

v1.9.2_hw1_universal

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

Generic/aliexpress

### System used

Nintendo 64

### Bluetooth controller brand & name

N/A

### What is problem? (only list ONE problem per report)

I just tried updating my blueretro adapter from 1.9.1 to 1.9.2 (using both bluetooth and USB) and it refuses to boot. After flashing 1.9.2 and resetting it, the blue light blinks once every 10s or so instead of doing the normal pulse when waiting to connect to a controller.

I am able to see see the adapter pop up as a bluetooth device on blueretro.io, but it refuses to connect 100% of the time. Downgrading back to 1.9.1 via USB fixes the issue.

I also tried with the N64-specific firmware, and this seems to work fine. It seems to be an issue only with the universal firmware.

### What did you expect to happen?

Adapter works with 1.9.2 universal firmware.

darthcloud, 2024-04-11:
Universal v1.9.2 work fine for me on my own adapter.

If possible with your adapter try to get the serial logs:
https://github.com/darthcloud/BlueRetro/wiki/Getting-BlueRetro-debug-logs-via-Serial-port-Windows-10

asweintraub, 2024-04-12:
Just got a chance to try this out. I get the following on loop every 10 or so seconds.
[blueretro logs.txt](https://github.com/darthcloud/BlueRetro/files/14961469/blueretro.logs.txt)

darthcloud, 2024-04-14:
v1.9.2 now use the NVS partition, I guess somehow that area on your specific device got something into it that make the adapter crash when running with v1.9.2.

Try to reflash v1.9.2 and add the ota.bin 2 additional time at 0x9000 & 0xB000 to reset the NVS partition like so:
[image]

Full doc:
https://github.com/darthcloud/BlueRetro/wiki/Flashing-firmware-Windows-10

asweintraub, 2024-04-14:
Thank you! Just tried flashing those partitions as well, and I'm getting the exact same error. Do I just have a bad device, or could it be something else?

[image]
[blueretro logs2.txt](https://github.com/darthcloud/BlueRetro/files/14971559/blueretro.logs2.txt)

darthcloud, 2024-04-14:
Probably not if v1.9.1 work fine, I'll look into the change of v1.9.2 again. I might give you a few different build to test later this week.

v1.9.2 universal work fine for my on my N64 so this quite strange.

darthcloud, 2024-04-14:
Try to flash the ota bin at 0x310000 also.

On first power on it won't work right away, give it a good 1 minute to recreate it's partition (status led will be off) when its done the LED will start pulsing as usual.

Power off and power on again let me know if that work.

asweintraub, 2024-04-14:
> v1.9.2 universal work fine for me on my N64 so this quite strange.

It doesn't actually seem to matter what console I try to use. I've tried booting the BlueRetro with an N64 cable, Wii Ext cable, or no cable at all, and I get the same issue with all of them.

> Try to flash the ota bin at 0x310000 also.

I just tried, but the blue status led was on during the entire time it was formatting. Once it was finished, it was back to the same error message as before.

Thanks again for all of your help with this!

[blueretro logs 3.txt](https://github.com/darthcloud/BlueRetro/files/14971818/blueretro.logs.3.txt)

asweintraub, 2024-04-15:
That worked!

Thank you so much for all of the help, and let me know if you need me to run any more tests on this.

darthcloud, 2024-04-24:
Fixed in v24.04

closed 2024-04-24 by darthcloud