darthcloud/BlueRetro #954: Unable to Upgrade Firmware From 1.8.x on RetroScaler N64 Adapter

FairyKid64, 2024-04-09:
### BlueRetro firmware version

1.8.3 (or perhaps 1.8.2)

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler

### System used

Nintendo 64

### What is problem? (only list ONE problem per report)

I updated my [RetroScaler BlueRetro adapter](https://www.aliexpress.us/item/3256805529834875.html?spm=a2g0o.order_list.order_list_main.10.572618021TGLQv&gatewayAdapt=glo2usa) last year to I think 1.8.3 (or 1.8.2?), and now I've been trying to connect to update the firmware or enter the advanced config page at blueretro.io, and it doesn't connect.  The log output just freezes on "Getting BlueRetro Service...".

I don't know if this is related, but the update and advanced config pages appear to have broken around the same time as [this website update](https://github.com/darthcloud/BlueRetro/discussions/827#discussioncomment-7572402) was made.

It's also worth mentioning that not all pages on the blueretro.io site are broken.  For example - the [BlueRetro N64 controller pak manager page](https://blueretro.io/n64_ctrlpak.html) does still work.  Also, I can still connect controllers to the adapter to play games on the Nintendo 64 - I just can't update or change the configuration.

### What did you expect to happen?

I expected to be able to use all the features of the blueretro.io website like update to the latest firmware and change the advanced configuration like I used to be able to.

### Attach files like logs or Bluetooth traces here

**Requesting Bluetooth Device...
Connecting to GATT Server...
Getting BlueRetro Service...**

_Then it just freezes there and never updates.  If I turn off the N64 while it's "Getting BlueRetro Service", then it prints out this message:_

**app_ver_is_hw2: false app_name_is_hw2: false
Init Cfg DOM...
\> Bluetooth Device disconnected**

_Here it would appear that website thinks this is a hardware 2 adapter, but it is hardware 1, and the last time I was able to update it last year, I updated it with the hardware 1 firmware._

darthcloud, 2024-04-11:
Is this ota page work better?

https://darthcloud.github.io/BlueRetroWebCfg_Beta/ota.html

FairyKid64, 2024-04-11:
> Is this ota page work better?
> 
> https://darthcloud.github.io/BlueRetroWebCfg_Beta/ota.html

Hi DarthCloud - thanks for responding to this issue.  I just tried this beta ota site on my Mac using Chrome as well as on my Android phone using Chrome, and it's still not working.

darthcloud, 2024-04-11:
You mentioned the n64 page work, what firmware does that page report?

lpuzzello, 2024-04-13:
Facing the same issue and tried some troubleshooting from my end. Hope this is useful:

- Different Bluetooth dongle (4.0 and 5.0), Computer and also my mobile close to the N64
- OTA FW update, System manager, Preset configs, Advance config.
- N64 controller pak manager page works and I can even format the memory pak.
- I factory reseted and following information is seen Sys Manager Page "BlueRetro_N64_B57E (10:97:bd:93:b5:7e) [1]"

The interesting part is that the behavior/error messages changes depending of the page/operation:

- Advanced Config page

 Reading App version...
 Reading Api version...
 Requesting Bluetooth Device...
 Connecting to GATT Server...
 Getting BlueRetro Service...
 ABI version: 2
 Init Cfg DOM...
 Get Global Config CHRC...
 Reading Global Config...
 Argh! NotSupportedError: GATT operation failed for unknown reason.
 > Bluetooth Device disconnected

- Presets, OTA FW update

 Requesting Bluetooth Device...
 Connecting to GATT Server...
 Getting BlueRetro Service...
 app_ver_is_hw2: false app_name_is_hw2: false
 Init Cfg DOM...
 > Bluetooth Device disconnected

- N64 controller pak manager and System Manager
  
   Works and I can even format the Memory Pak in the Pak Manager

If needed I can perform more tests and provide more information.

Thanks !

FairyKid64, 2024-04-13:
> You mentioned the n64 page work, what firmware does that page report?

It actually doesn't show a firmware version on the Memory Pak page.  But from what I remember, I think I last updated to 1.8.3 or 1.8.2.  This is what it shows on the Memory Pak page:

Please consult the documentation found on the wiki:
[7.1 - N64 controller pak manager page](https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#71---n64-controller-pak-manager-page)

Connected to: BlueRetro_N64_AD42 (10:97:bd:93:ad:42) [1]

Download latest FW v1.9.2 from [itch.io](https://darthcloud.itch.io/blueretro)
Select BlueRetro controller pak bank:

darthcloud, 2024-04-13:
Ok I understand now, its running a custom FW from retroscaler and they didn't populate the FW version field properly. So the OTA probably dont like that.

I'll see what I can do.

FairyKid64, 2024-04-13:
> You mentioned the n64 page work, what firmware does that page report?

It actually doesn't show a firmware version on the Memory Pak page.  But from what I remember, I think I last updated to 1.8.3 or 1.8.2.  This is what it shows on the Memory Pak page:

Please consult the documentation found on the wiki:
[7.1 - N64 controller pak manager page](https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#71---n64-controller-pak-manager-page)

Connected to: BlueRetro_N64_AD42 (10:97:bd:93:ad:42) [1]

Download latest FW v1.9.2 from [itch.io](https://darthcloud.itch.io/blueretro)
Select BlueRetro controller pak bank:

> Ok I understand now, its running a custom FW from retroscaler and they didn't populate the FW version field properly. So the OTA probably dont like that.
> 
> I'll see what I can do.

Ah, I see.  Yeah, the last time I updated I'm pretty sure I downloaded the official firmware, but it was a few months ago, so maybe I'm remembering incorrectly.  Although I also tried doing the factory reset, so perhaps that put the adapter back to the custom RetroScaler firmware.  Thanks for looking into this.

lpuzzello, 2024-04-14:
> Ok I understand now, its running a custom FW from retroscaler and they didn't populate the FW version field properly. So the OTA probably dont like that.
> 
> I'll see what I can do.

Thank you DarthCloud.
I was checking the code of the OTA page and found this line:
[image]

I changed to '1' in my local cache and could update sucessfully to 1.92
[image].

darthcloud, 2024-04-14:
The latest beta page should fix the issue:
https://darthcloud.github.io/BlueRetroWebCfg_Beta/ota.html

Let me know if that work.

FairyKid64, 2024-04-14:
> The latest beta page should fix the issue: https://darthcloud.github.io/BlueRetroWebCfg_Beta/ota.html
> 
> Let me know if that work.

Yes - that beta page worked!  I was able to update to 1.9.2.  I am also now able to enter the advanced configuration page.  Thank you DarthCloud!  And thank you Ipuzzello for helping debug!

darthcloud, 2024-04-14:
Awesome, I updated the regular page too

closed 2024-04-14 by darthcloud