darthcloud/BlueRetro #955: 8bitdo N30 Mod Kit and N30 Controller issues

s3anami, 2024-04-11:
### BlueRetro firmware version

1.9.2

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

I wired an ESP32 connected to NESTang tang20k

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

8bitdo N30

### What is problem? (only list ONE problem per report)

I am trying to use Blueretro and 8bitdo N30 devices on a NESTang.
The NESTang is taking in PS2 style inputs as controllers

It seems like it was recommended to use the 8btido N30 modkit in Windows mode, however when I try it it does not work and will only work properly if in Switch or Apple mode. Do not know if this is intended behavior. In Windows mode the mapping is off and only certain buttons will work.

For the normal 8btido N30 devices I have tried the old and newer version of the controller and neither will map the D-pad correctly. I have tried in advanced config to use the auto mapping and have also tried using the NES source option to setting joystick buttons to the dpad but it doesn't seem to do anything. Possibly I am not configuring this properly. The A and B buttons seem to work properly though.

### What did you expect to happen?

Buttons to map

darthcloud, 2024-04-14:
You need to use Xinput mode and also configure the N30 dpad to be a dpad (it's a joystick by default)

See:
https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#73---d-pad-as-joystick-or-d-pad-configuration

Make sure to put your blueretro config back to default aswell:
You can load "default gamepad" preset:
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#33---example-preset-load

darthcloud, 2024-10-06:
Feel free to reopen if you got more info

closed 2024-10-06 by darthcloud