darthcloud/BlueRetro #956: RT / LT trigger bug (Source Xbox, Dest. NGC)

vincbeaulieu, 2024-04-13:
### BlueRetro firmware version

1.9.2

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

BlueRetro

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Xbox One / Series X|S

### What is problem? (only list ONE problem per report)

Using 1 connected Xbox controller.
Trigger buttons not working unless both Digital and Analog RT sources are mapped to the same destination RT in preset. Same for LT.

Using 2 connected Xbox controllers.
Using the same preset for output device #2, where both Digital and Analog RT sources are mapped to the same destination RT, and the same for LT. We now have to press on both controller RT buttons for the player #2 to perform the action liked with RT.

### What did you expect to happen?

trigger button should be one-to-one mapped between input devices and destination port.

vincbeaulieu, 2024-04-13:
I've tried the following #753. The fix is effective for one connected controller but stops working when a second controller is connected (where you have to press on both controllers same trigger for it to work).

darthcloud, 2024-04-14:
Yes there is indeed something weird going on with the trigger digital end when using more than one controller.
I will look into it.

But for the part that you need to map the analog and digital on same button, this is normal.
The Xbox (and all modern controller) don't have the same type of trigger than GC (which is 2 buttons in one).
The only controller that wouldn't require a special mapping in BlueRetro is the Wii classic OG controller which got the exact same type of trigger.

It really depend on which game you use, on some games you will want to split the analog and digital part to different button but most of the time you will want it on same trigger. This is why there is a preset for that:
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#33---example-preset-load

Also another important note for Xbox controller is that you need to press a digital button like A first a couple time before touching the trigger and joystick to allow blueretro to calibrate them properly. This is noted here:
https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#4---xbox-pairing-guide

darthcloud, 2024-04-14:
This beta should fix it:
https://github.com/darthcloud/BlueRetro/actions/runs/8680779369

darthcloud, 2024-04-24:
Fixed in v24.04

closed 2024-04-24 by darthcloud