darthcloud/BlueRetro #958: Stadia controller L trigger sends both Z and C-up on N64

asweintraub, 2024-04-15:
### BlueRetro firmware version

v1.9.1_hw1_universal and v24.04_beta_universal_hw1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

blue-retro.com

### System used

Nintendo 64

### Bluetooth controller brand & name

Google Stadia

### What is problem? (only list ONE problem per report)

Using the left trigger on the Stadia controller activates both C-up and Z in N64 games. When pressing the trigger very slightly, C-up activates immediately, but Z has slightly more of a dead zone before activating. The right trigger on the Stadia controller only activates Z on the N64, which is what I would expect.

It seems as if the left trigger controller axis could be getting mapped to the RY axis (in addition to the Z button), but I haven't confirmed this.

### What did you expect to happen?

The left trigger on the Stadia controller only activates Z and doesn't also press C-up.

asweintraub, 2024-04-19:
From the duped bug:

Pressing the menu button on the Stadia controller activates both the Start button and C-right on the N64 (confirmed with a controller test rom).

darthcloud, 2024-04-19:
Please try this beta:
https://github.com/darthcloud/BlueRetro/actions/runs/8757581399

I did see the L button issue in my n64 rom test but not the start button one.

Let me know if it fix one or both of the issues.

asweintraub, 2024-04-19:
Yup, just tried this out and all the buttons work properly now. Thank you for the quick fix!

darthcloud, 2024-04-19:
I handle the closing :P

asweintraub, 2024-04-19:
Oops, sorry about that!

darthcloud, 2024-04-24:
Fixed in v24.04

closed 2024-04-24 by darthcloud