darthcloud/BlueRetro #960: 3DO Fz-10 (Anvil) adapter hasn't input

darthcloud, 2024-04-15:
### Discussed in https://github.com/darthcloud/BlueRetro/discussions/868

<div type='discussions-op-text'>

<sup>Originally posted by **CarloScalabrin** December 23, 2023</sup>
Hello everyone, I'm facing some problems to get Blureretro Core to work on the 3DO FZ-10 with Anvil chip. I am using pmgducati's PCB design. On Fz-1 and Fz-10 with DSP Clio it works normally, however on Fz-10 models with DSP Anvil, it does not work.  The core has power, the control pairs normally, but there is no button input. Any tips/suggestions?</div>

towerpc, 2024-09-30:
I have Anvil 3DO, and have to problem. My diy BlueRetro HW2 not work. I look at oscilloscope data bus, and seen that my sontroller is work how second (№2), without first. I deleted all contoller and bind controller, againe my controller have second position on oscilloscope signal.

darthcloud, 2024-10-06:
@towerpc Can you upload screenshot of your scope capture? Try to get the beginning of the frame.

BuzzBuzzards, 2024-12-09:
Have the same (or similar) issue here. I also have a FZ-10 with Anvil chip. And what puzzling me that some games (such as Cannon Fodder) allow me to use BlueRetro in the menus, but not during the gameplay. The only exception, when game allowed to regain my control back is than my squad takes damage. 

The same situation goes to the Return Fire, but with a slight difference. Main menu works ok and shows, that there is two controllers. BT controller, connected as a second one, has no input registered no matter which buttons I press. When I choose single player campaign - no input from either of BT controllers being registered. 

When I assign second input as a keyboard - Return Fire shows only one controller, but also have no luck with playing single player campaign. 

Also, my 3DO has a USB ODE installed. And in Fixel's boot menu BlueRetro also works perfectly. But! If I daisy-chain my regular 3DO controllers and try to use the second one in the ODE menu - it didn't work. So, I assume that it my setup BlueRetro works neither as normal first nor second controller. 

HomeBrew app for controller testing also shows no inputs being registred. 

At some point I thought that the issue is somehow connected with a lack of power, so I powered my esp32 via USB. Had the same result, but I managed to collect some logs. 
[blueretro_3do.log](https://github.com/user-attachments/files/18060236/blueretro_3do.log)

FW: v25.01-beta-3-g51cf1dc
HW: AliExpress BlueRetro Core + DIY Cable
Controller: 8BitDo M30 and 8BitDo SN30 Pro 2

darthcloud, 2025-01-28:
Turn out this issue is related to the level shifter used.

AHC125 wont work

you needs 74AHCT1G125 or 74AHCT125

Note the **AHCT** you need all 4 letters.

I debugged one HB adapter and swapping the AHC125 for AHCT125 fixed the issue with the Anvil instently!!

closed 2025-01-28 by darthcloud

BuzzBuzzards, 2025-01-29:
[image]
The problem is that I'm using SN74AHCT125N. Do you know what is the exact difference between those chips? I'm not good enough to tell that only by datasheets. I want to check if my chips counterfeit or not.

BuzzBuzzards, 2025-01-29:
Ahh, found solder bridge between 3DO controller pins. Thanks for your help!

Metro-City, 2025-03-23:
i have a 3DO Fz-10, dongle need to fix the mapping of the dual shock 4 since an action button is on the options button