darthcloud/BlueRetro #961: [1.8.4] Brook X One adapter does not work in Switch and PS4 mode

darthcloud, 2024-04-15:
### Discussed in https://github.com/darthcloud/BlueRetro/discussions/749

<div type='discussions-op-text'>

<sup>Originally posted by **Invictaz** August 19, 2023</sup>
Connected the Brook X one Adapter to my Xbox One 1708 controller. 
Brook adapter set to Nintendo Switch Pro controller mode. 
Does not want to pair with Blueretro PSUnoRetro Adapter.

HCI log captured via ubuntu 

Nintendo Switch setting

[hcitrace.zip](https://github.com/darthcloud/BlueRetro/files/12387004/hcitrace.zip)

[~hcitrace.txt](https://github.com/darthcloud/BlueRetro/files/12387002/hcitrace.txt)

and in PS4 mode it does connect and the PSUnoRetro light goes off, but there is no button input.

[~hcitrace2.txt](https://github.com/darthcloud/BlueRetro/files/12387011/hcitrace2.txt)
[hcitrace2.zip](https://github.com/darthcloud/BlueRetro/files/12387012/hcitrace2.zip)

</div>

Invictaz, 2024-04-15:
You need any more logs from me @darthcloud ?

darthcloud, 2024-04-15:
No, I just need time to look into it.

darthcloud, 2025-04-16:
Feel free reopen once you got the trace.

closed 2025-04-16 by darthcloud