darthcloud/BlueRetro #976: NES Multitap random button Presses on NES Frontloader

raschinger, 2024-04-23:
### BlueRetro firmware version

1.9.2

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

Self made 2 player dongle 

### System used

Nintendo Famicom NES

### Bluetooth controller brand & name

Switch, 8BitDo, Xbox series controller

### What is problem? (only list ONE problem per report)

Recently built my own 2 port Dongle for the NES. I've tested the Multitap feature. And found the following behavior. Once the 3rd controller is connected to BlueRetro. I've observed random button presses on Player 2.

https://github.com/darthcloud/BlueRetro/assets/52976332/aec6bfb7-6cee-4ddf-a61e-ba4063c97c50

### What did you expect to happen?

I expect to get the Multitap feature working.

darthcloud, 2024-04-23:
Please provide schematic and BOM for me to review

bixxewoscht, 2024-04-26:
I just tested this with NES Hub V2.1 (same one you have @darthcloud.
Same behaviour, 2 controller without four-score activated in the web-config work fine.
With four-score activated I experience the same inputs on controller 2. Both with only 2 controllers connected and also with 4 controllers connected.

darthcloud, 2024-04-26:
On my AVFC I only get a few random press on P2 A button that's all using v24.04.

I actually just realized I got no way to run ROM on a NES. I don't have a FC to NES adapter (and none of the games that include one).

I might try to get a Gyromite cartridge, until then I can't really easily test.