darthcloud/BlueRetro #985: N64&NSO controller - Rumble doesn't work at all in game

JayBee-git, 2024-05-02:
### BlueRetro firmware version

1.9.2

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

8bitMods

### System used

Nintendo 64

### Bluetooth controller brand & name

NSO N64 controller

### What is problem? (only list ONE problem per report)

The rumble doesn't work at all in game. When I press the home button I can feel 2 different rumbles pulses, but the rumble never works in game. I tried every mode in the web advance config, and also updated to the latest 24.04, no changes. 
I looked up online and here, and couldn't find any info either.
When I connect a real controller with rumble pack instead of the adapter, the rumble work instantly.

### What did you expect to happen?

I expected the rumble to work once I enable it from the web UI.

darthcloud, 2024-05-03:
BlueRetro only support D mode currently, and in that mode 8bitdo decided to not support rumble.

darthcloud, 2024-05-03:
Are you using a real NSO from Nintendo or the 8bitdo mod kit?

JayBee-git, 2024-05-03:
A real NSO controller, connected to the 8BitMods blueretro adapter.

JayBee-git, 2024-05-03:
I also updated the controller just in case, and tested on the official NSO service, rumble works normally there. 
When pushing the mode button, I also get the rumble pulses so I don't think there is an issue with the controller.
It seems like it's either not recognized by the game or not triggering the rumble.

darthcloud, 2024-05-03:
Did you enable rumble in the config?
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#26---example-enable-rumble

JayBee-git, 2024-05-03:
As I said, yes I did. Several times. I was very careful and checked many times from your wiki if I was doing things the right way. Everything is set properly. It's just not receiving/triggering rumble.
Also did factory resets, upgraded/downgraded to no avail.
Is the rumble function tested as working on the official NSO controller with the lastest firmware(s)?

JayBee-git, 2024-05-26:
I downgraded the firmware to the oldest one available in the download page (1.8.3), but same issue, the rumble doesn't work at all in game, no matter the mode selected in the advanced config page.
I can see old posts and videos showing the NSO rumble working with far older firmwares, did the rumble support brake at some point?
Can anybody confirm if it's working for them?

Since I do get short rumble pulses on mode change when pressing the home button, and otherwise all buttons work, I don't think there is an issue with the controller or the communication with it, it seems to be a problem on the blueretro adapter side.

I'm willing to test out other things if necessary.

darthcloud, 2024-05-29:
So base on the discord this issue only happen when using mister.

Yet none of the post here mention the use of mister.

I don't have one so it's not something I can debug.

JayBee-git, 2024-05-29:
My bad, it wasn't working either on the real N64 before, for convenience sake I did the rest of the tests on the Mister. The weird part is that even on the Mister the rumble works perfectly when using a homebrew, just not with native games, whereas og rumble pak work everywhere.
Can I help you with the debugging on that one, or is it a lost cause?

darthcloud, 2025-01-24:
Can you try again in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)?

CJ5392, 2025-01-25:
I can confirm with v25.01 rumble does not work with the NSO N64 controller. Tried changing output to rumble, memory or both and when switching rumble does not work

MadBullBunny, 2025-05-19:
I'll add to this issue, using a brawler64 nso controller and some games detect rumble while others do not. I've tried everything and some games like conkers bfd won't detect the rumble. Both on the latest firmware possible.