darthcloud/BlueRetro #995: Can't play Golden Axe The Duel on Saturn due to Gamepad type

DarthMotzkus, 2024-05-11:
### BlueRetro firmware version

v24.04

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler

### System used

SEGA Saturn

### Bluetooth controller brand & name

8Bitdo M30 Bluetooth Controller

### What is problem? (only list ONE problem per report)

Games that have no compatibility with 3D analog controller reset after SEGA logo, like Golden Axe The Duel.

### What did you expect to happen?

Since some Saturn games do not support 3D Analog Controller, we need an Output Config Mode of a standard Saturn gamepad, not the 3d controller emulated with dpad setted.
This happen with original controller too. Only with the standard controller Golden Axe The Duel work properly.
Already tested with GamepadAlt mode, but this only emulate the 3d controller with analog setted to on. No input is recognized, even the game do not reset after Sega logo, but it reset if you change the controller mode to Dpad (button - on M30).

darthcloud, 2024-05-11:
Yeah this intended.

Supporting the regular controller protocol is likely not possible using a SoC alone.

So BlueRetro got the same limitation as a official 3D controller.

DarthMotzkus, 2024-05-11:
What you mean by SoC? Isn't there any workaround to play those games which is not compatible with 3D controller?

marcomalachias, 2024-05-12:
> ### BlueRetro firmware version
> v24.04
>
> ### BlueRetro firmware specification
> HW1
>
> ### BlueRetro firmware variant
> System specific
>
> ### BlueRetro hardware type
> External adapter dongle (1 port only) [...]

Yeah, I can confirm that and couldn't find a solution yet. My Saturn model is a Brazilian Tectoy and I tested the game with 8bitdo m30 gamepad, dualsense, dualshock 4 and Xbox controller. I wonder if there is any workaround for this issue.

lamprey13, 2024-05-15:
That is a bummer. I was just thinking of getting some BlueRetro receivers for my Saturn as the offical Retro-Bit Saturn Controller/Receiver act the same way. Even more bothersome, is they seem to cause issues with the Saroo's Menu system. Not sure if a BlueRetro solution would be any better or not with the Saroo?