darthcloud/BlueRetro #1000: Right Trigger wont work

curtcooll, 2024-05-14:
### BlueRetro firmware version

24.04

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

8bitmod

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Xbox series x 

### What is problem? (only list ONE problem per report)

The right trigger does not function even if i try a different button on the xbox controller still wont function Left trigger works fine

### What did you expect to happen?

Expected it to work

Wild-Dog, 2024-10-01:
Same issue on ps3 controller, can assign left trigger input to R2 source and it works just fine. Right trigger input itself isn't working.

darthcloud, 2024-10-06:
OP Feel free to reopen if you got more info.

Wild-Dog better open a new issue rather than post in someone else report.

closed 2024-10-06 by darthcloud