darthcloud/BlueRetro #1001: Some games don't recognize the virtual VMU

gorgyrip, 2024-05-15:
### BlueRetro firmware version

24.04

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

me

### System used

SEGA Dreamcast

### What is problem? (only list ONE problem per report)

Some games like Carrier and Resident Evil 2 don't recognize the virtual VMU. It works for other games and I can also see the saves in the console menu.

### What did you expect to happen?

I was hopping to fix this issue/

### Attach files like logs or Bluetooth traces here

...

darkymtp, 2024-06-23:
Hello.

Same here..

IMMORTAL1981, 2024-08-12:
I ma having this same problem.
Soul Reaver and Resident Evil 2.
Both games do not recognize the Virtual VMU.

My Dreamcast has the Virtual VMU formated and installed.

What other gamer have you guys tried?

Cheers!

darkymtp, 2024-10-06:
Carrier still says "No VM found" with 24.10 firmware.

EatCrunch, 2024-11-03:
Firmware 24.10 here i have same problem with soulreaver too and bust a move 4

darkymtp, 2025-01-24:
Version 25.01
Same problem.
Soul Reaver : No error message but saving is not available.
Bust a move 4 : "Save failed"

darkymtp, 2025-04-29:
v25.04 same problem.

Ashtray3250, 2025-05-06:
Resident Evil: Code Veronica also does not recognize the virtual VMU, after saving, if you try to load "bad save error" sad, I hope you fix this problem

darkymtp, 2025-05-06:
I had no problem with Code Veronica X and the virtual VMU.

darthcloud, 2025-05-18:
Try this beta it should make the games mentionned in this thread see the emulated VMU.

[v25.07-beta-6-g3852a4d5_hw1_dreamcast.zip](https://github.com/user-attachments/files/20273012/v25.07-beta-6-g3852a4d5_hw1_dreamcast.zip)

znam1980, 2025-07-11:
Hi
I have similar problems in the game Carrier
on 25.04 and below the game simply does not see the VMU
I tried the beta firmware 25.07 beta 11 and the game already sees the memory card but the message "A controller has been removed or the memory card is being checked" constantly blinks
it is very difficult to play
plus on this beta I have problems in the game Rayman 2, there it also periodically writes that the gamepad is disabled, although this game works fine on 25.04
maybe it is worth trying some new beta? , or should I wait for a new full firmware

znam1980, 2025-07-13:
[image]
I checked the firmware number a couple of times and tried to reflash a couple of times
the screenshot shows the firmware number
The VMU in Carrier essentially works, but the message that the gamepad is disabled constantly flashes
also the message that the gamepad is disabled is in Rayman 2 and ECCO the Dolphin, although on firmware 25.04 everything is fine
after flashing to beta, I reset the settings to factory settings and re-formatted the VMU on the console
I will also note that in the game KISS Psycho Circus the VMU is not detected at all, even on beta firmware

znam1980, 2025-07-13:
If needed I can make a short video with problematic games

darthcloud, 2025-07-13:
@znam1980 Please open a new issue, I think there is something different going on in your setup.

Please list a much detail in it. (wireless controller, blueretro model/type/brand, mods present on dreamcast if any)

I'm going to close this one since all games listed here work for me now.

closed 2025-07-13 by darthcloud

Ashtray3250, 2025-08-05:
in Resident Evil 2 all the buttons on 8bitdo are mixed up and saving doesn't work