darthcloud/BlueRetro #1002: BSP-Y02 issues after restart console

sidneivl, 2024-05-15:
### BlueRetro firmware version

24.04 saturn and parallel 1p

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

8bitDo

### System used

SEGA Saturn

### Bluetooth controller brand & name

BSP-Y02

### What is problem? (only list ONE problem per report)

I tried pairing my BSP-Y02 using the X+HOME combination for the Android connection (other connection types don't work).

 All buttons worked as expected, but the directional buttons weren't working.

So I've noticed that is possible to map the controller, and I've added LX axis in SRC to D-pad axis on DEST in advanced config.

[image]

If I make this configuration first and then enter the game, the controller works perfectly. I can play any game without problems.

However, if I turn off the console and turn it back on, the directional controls lose the configuration and respond in a strange way:

https://github.com/darthcloud/BlueRetro/assets/5385276/cd7807d4-6e2e-49fe-a93c-0f28aaeea931

This issue also happens with BlueRetro for Neo Geo MVS.

I've tried changing the connection and the mapping, but the directional configuration always gets lost after turning on the console again.

 Is there something I can do to solve this, or is it a real bug?

Regards

### What did you expect to happen?

I want to configure my BSP-Y02 controller to work with BlueRetro for Saturn and Neo Geo MVS games, but I need the configuration to persist after I turn off and on the console again.

SpotBRz, 2025-06-11:
Friend, did you find any solution to this problem? I also have the same problem with Saturn, Mega Drive and Neogeo

sidneivl, 2025-06-12:
> Friend, did you find any solution to this problem? I also have the same problem with Saturn, Mega Drive and Neogeo

No, I ended up selling the Saturn and the BlueRetro because I bought a MiSTer FPGA.