darthcloud/BlueRetro #1004: Virtual VMU isn't detected unless real controller is also connected

raygan, 2024-05-16:
### BlueRetro firmware version

24.04

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

AliExpress seller

### System used

SEGA Dreamcast

### Bluetooth controller brand & name

Sony DualShock 4

### What is problem? (only list ONE problem per report)

I found that with the [Napple Tale English translation](https://www.romhacking.net/translations/5091/), the game can save to the virtual VMU perfectly fine, but on the main menu screen it won't allow me to choose Continue to load an existing save unless a real controller/VMU is connected. 

If a real VMU is connected to any port, it passes whatever check is happening at the menu, and I can access the save-loading screen. On that screen, the virtual VMU _is_ recognized and I can continue my save from the virtual VMU.

### What did you expect to happen?

I expect the game to detect the presence of the virtual VMU

darthcloud, 2025-05-17:
That game got a bug! Unlike most games it doesn't like seeing a device that support only memory. It really want to see a VMU that support memory, clock & LCD.

I made the change to report all the features like a real VMU and the game look happy now.
Try this:

[v25.07-beta-6-g3852a4d5_hw1_dreamcast.zip](https://github.com/user-attachments/files/20266838/v25.07-beta-6-g3852a4d5_hw1_dreamcast.zip)

darthcloud, 2025-07-13:
Will be fixed in upcoming v25.07

closed 2025-07-13 by darthcloud