darthcloud/BlueRetro #1009: Castlevania: Symphony of the Night in Saturn

velho-VL, 2024-05-21:
### BlueRetro firmware version

BlueRetro_SATURN_B14A (a8:42:e3:a8:b1:4a) [v1.7.3 psx_external]

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

HONGYANG Phone Repair Parts Store

### System used

SEGA Saturn

### Bluetooth controller brand & name

8bitdo M30

### What is problem? (only list ONE problem per report)

I'm using an M30 on the Sega Saturn. All games seem to work fine except for Castlevania: Symphony of the Night. The M30 simply doesn't work with this game. I tried to find something about it, but I didn't find anything.

### What did you expect to happen?

I would like the M30 to be detected in Castlevania: Symphony of the Night.

velho-VL, 2024-05-21:
I tested with other controllers on the Blue Retro and the problem persists. This issue doesn't occur with a regular controller (without blueretro). I can't understand how an input system might not work in a specific game. Could it be that there is some type of saturn controller that is incompatible with this game?

rykotxet, 2024-08-14:
Hi. I don't own a Saturn, but in the PlayStation version of this game, playing the game with a PS2, an external BlueRetro and a DS4, the controller doesn't work unless you push the toggle analog button (PS/Home button on DS4 controller). Maybe it's the same case in Saturn, idk,

haydnkilyan, 2025-01-17:
I had this issue when I had my adapter set up as a multi tap - all other games worked but SotN only registered my M30 when the adapter was configured to behave like a regular single controller not a multi tap