darthcloud/BlueRetro #1015: Analog Trigger Not Working On GameCube

littlemisskittn, 2024-05-27:
### BlueRetro firmware version

24.04

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install modkit

### Manufacturer

Laserbear

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Xbox Series X OEM

### What is problem? (only list ONE problem per report)

Right Trigger is not working properly in Wario World despite the correct preset being loaded. Cannot perform slam move which is necessary to progress. 

### What did you expect to happen?

To press down the right trigger with A to do a slam move.

littlemisskittn, 2024-05-28:
Also thinking it’s an issue with mapping this game in default settings because pressing LT, LB, or RB performs the same action as L in the main game (sucking up coins as per the manual)

littlemisskittn, 2024-05-28:
Reset adapter and it went back to 1.9.2. Tried it again and still doesn’t work. LT, LB, and RB still map as the L trigger.

darthcloud, 2024-05-29:
You got to use v24.04, all other version the trigger a broken

You also need to follow those two step make it work:

You need to use the merged preset:
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#33---example-preset-load

Also another important note for Xbox controller is that you need to press a digital button like A first a couple time before touching the triggers and joysticks to allow blueretro to calibrate them properly. This is noted here:
https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#4---xbox-pairing-guide

littlemisskittn, 2024-05-30:
I did that and it still doesn’t work. As I also said, the LB, RB, and LT buttons do the same thing which is sucking up coins while the RT doesn’t do anything. I followed the directions to a T, even reset the device, reinstalled the current firmware, and followed the guide again. Still won’t work properly. The RT works fine in _other_ games that use analog triggers that I tried (Wrestlemania X8 uses them for the arena selection, Metal Gear uses them for inventory, though only the analog part works all the time, sometimes the digital part doesn’t work as well which is quick equip). It’s not anything I’m personally doing wrong as I’ve done all the steps right, numerous times.

darthcloud, 2024-05-30:
https://youtube.com/shorts/Ip0m7lI2HsQ?feature=share

It work fine for me.

Did you update the Xbox controller to latest FW as noted in the pairing guide docs?

Take a screenshot of the mapping config you see at:
https://blueretro.io/advance.html

Also take a screenshot of your Xbox FW version in the Win10 Xbox accessories app.

littlemisskittn, 2024-05-30:
Here’s the snapshots from the app:

littlemisskittn, 2024-05-30:
and here's the snapshot of the Xbox Accessories app on my Windows 11 PC

twinfami, 2024-05-30:
In case it didn't come across via email:

https://github.com/darthcloud/BlueRetro/assets/67311558/6e21b055-a430-4266-9166-ec8115073a3e

[image]
[image]
[image]
[image]
[image]
[image]

littlemisskittn, 2024-05-30:
[image]

littlemisskittn, 2024-05-30:
https://youtube.com/shorts/IligZKC1pfo?si=uH20gdc8jjImDS6m

littlemisskittn, 2024-05-30:
Just also tried smash. That works fine with the analog and digital merged. So that’s like the fourth game I tried that used the analog and digital trigger that worked. It’s just _Wario World_ not working and I think that’s down to how the game sees it’s mapped. There’s no option to change in game as like I said, the LT, LB, and RB buttons in Wario World act as the L trigger for some reason and will not detect the RT at ALL. You have everything in front of you so unless it’s a problem with Laserbear’s board (which I doubt as the other games wouldn’t work neither), I don’t know what else could be the problem when everything else is working as intended.

darthcloud, 2024-05-31:
Your Xbox ctrl got same FW as mine, the config look fine, I see 2 different version of FW used, just make sure you stick to v24.04 to make troubleshoot easier.

On gamecube the triggers are 2 button in one. So for games that works its just because it's using the analog part, and for those that don't it because it use the digital end part.

So the merged presets is what adds the digital end at the end of the analog trigger most modern BT controller got.

The only thing I can think off is that maybe your controller trigger maximum isn't reaching as far as most other xbox controller.

Try to change the threshold to 75% in advance config to see if that help:
[image]

littlemisskittn, 2024-05-31:
I tried at that value and I’ve even had it lowered all the way to 0% for Digital and 25% for Analog. Only thing it does now is still do the same thing the other shoulder buttons do. No pound or dash. Mario Party 4 had a similar problem where I wasn’t able to move the page at all.

sushimehl, 2024-06-14:
I also got this issue with 8bitdo Pro2 Controller for this Game since a long time now... The merge trigger preset does not work for me, only the Controller mapping gets wrong again, every time I try it...

treyus30, 2024-06-30:
I'm going to 2nd and expand this issue. I'm on latest firmware and realized yesterday that neither my analog or digital triggers work for device flashed for GameCube, using Dual shock 4, in super smash bros brawl (I'm assuming Meele would have the same problem). So I can't shield or dodge.
Can't verify my config because I refuse to download Google spyware and I can't find the guide that shows you how to do it without the webhost.

haydnkilyan, 2024-09-22:
I have the same issue on warioware - latest firmware ☑️ merged trigger preset ☑️ all other games ok ☑️ wario world right trigger ❌ does not work - tried dual shock 4 and 8bitdo pro 2 controllers ❌❌

4a, 2024-10-31:
Just adding that this issue is also occuring for me on latest firmware (v24.10_hw2) using the 8bitdo Gbros Adapter with a real GCN controller on both the default gamepad preset, and the merged analog preset. L analog and L digital press are working fine, but R trigger seems to not be detected at all (you can test the digital press in the pause menu). The issue only exists in Wario World as far as I know.

Firmware: 24.10_hw2
Controller: 8bitdo Gbros with original GCN controller
Receiver: Laser Bear Internal v2
Preset: Default Gamepad / Merged Analog (Neither preset has R trigger detected at all in ONLY this game)

darthcloud, 2025-01-24:
I made a new way to debug issue like this in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01), could you try to get me a debug trace following theses instructions?
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

Thanks

4a, 2025-02-03:
> I made a new way to debug issue like this in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01), could you try to get me a debug trace following theses instructions? https://github.com/darthcloud/BlueRetro/wiki/Debug-trace
> 
> Thanks

Sorry for the delay on this. I tested out using the R trigger on the ground and while jumping, still no response. L trigger seemed to work perfectly.

Let me know if you need more info.

[br_debug_trace.zip](https://github.com/user-attachments/files/18649301/br_debug_trace.zip)

darthcloud, 2025-02-18:
I finaly reproduced this problem, It' worked for me because I was using a Wii with nintendont somehow.

I tried with an actual GC and swiss using the same adapter and I got the issue now.

Looking into it

darthcloud, 2025-02-18:
This beta should fix the issue:
https://github.com/darthcloud/BlueRetro/actions/runs/13382675228

sushimehl, 2025-02-18:
Can confirm it's working now. Thank you !

darthcloud, 2025-02-18:
Thanks for testing!, Will be fix in v25.04 later this spring.

closed 2025-02-18 by darthcloud

Jdbye, 2025-03-10:
Thank you for the fix. I got an internal blueretro mod about a month ago and this was one of the first games I tried, so was bummed the issue makes it pretty much unplayable. Funny coincidence that it was fixed only about 10 days later.