darthcloud/BlueRetro #1017: DS3 Vibration Delay v24.06-beta-22

ButtonBasher92, 2024-05-29:
### BlueRetro firmware version

v24.06-beta-22-ge7ba6d6_hw1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

PlayStation 3 Dualshock 3 CECHZC2U

### What is problem? (only list ONE problem per report)

There is a Delay with vibration with one of my DS3 controllers, model CECHZC2U (confirmed to be Original)
My other DS3 controllers, model CECHZC2E work just fine with the same BlueRetro adaptor.

### What did you expect to happen?

Vibrations to be synchronized with gameplay

darthcloud, 2024-05-30:
Yes I can reproduce with my CECHZC2U

My CECHZC2UB1 is fine however.

I'll see what I can do with that.

darthcloud, 2024-05-31:
This is better?
https://github.com/darthcloud/BlueRetro/actions/runs/9320769710

ButtonBasher92, 2024-06-02:
> This is better? https://github.com/darthcloud/BlueRetro/actions/runs/9320769710

That did the Trick, its working perfectly now. Amazing work thank you.

closed 2024-06-02 by ButtonBasher92