darthcloud/BlueRetro #1021: Random Left stick input DS3 v24.06-beta-24

ButtonBasher92, 2024-06-04:
### BlueRetro firmware version

v24.06-beta-24-gd0ef7fc_hw1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

PlayStation 3 Dualshock 3

### What is problem? (only list ONE problem per report)

Not sure if it's a bug or controller

But I get a random input from the left stick. it appears to be pressing up left very quicky at random intervals.
its noticeable in God of War PS2, character will quickly step to the left by itself and in final fantasy 8 PS1, in menus it will scroll to the left by itself along with character randomly stepping to the left.

What makes me think this is possibly a bug and not stick drift is when examining the controller in Padtest.elf in the RX and LX section L and R will flicker with a input very quickly. 
No apparent stick drift when testing the controller with various gamepad testers on PC
Exact same behavior happens with two DS3 controllers that I have CECHZC2U & CECHZC2E. My CECHZC2E A1 works fine though.
When same controllers are connected to a Brook wingman PS2 there are no issues.

### What did you expect to happen?

Stable accurate analogue stick input.

362053534, 2024-06-07:
I have the same problem.The character moved within 1 frame.
I do not touch the left stick. It's very random.No pattern.
I'm sure that is not stick drift.

Joseph-DiGiovanni, 2024-08-11:
Having the same issue with the methylene mod in my ps2. It happens on both a brand new CECHZC2U and the A1 revision contrary to OP. It does happen much less frequently on the A1 though.

Joseph-DiGiovanni, 2024-08-11:
I've downgraded from 24.06 beta to 1.9.2 and I thought the issue was entirely resolved but it just made it much less frequent. Downgraded to 1.9.0.2 (Before analog button support was added) and I haven't had an issue. This seems like a regression most likely caused by the addition of support for pressure sensitive button. Could you look into this @darthcloud?

Since the older firmwares for the Methylene mod aren't on the site I'll share them here for anyone in the same boat:
**THESE ARE FOR THE METHYLENE INTERNAL PS2 MOD ONLY**
[1.9.2](http://gusse.in/download/2219/?tmstv=1723404362)
[1.9.0.2](http://gusse.in/download/2225/?tmstv=1723404362)

darthcloud, 2024-09-14:
This build should fix it:
https://github.com/darthcloud/BlueRetro/actions/runs/10865078205

ButtonBasher92, 2024-09-16:
> This build should fix it: https://github.com/darthcloud/BlueRetro/actions/runs/10865078205

That seems to have done the trick!
Fixed, closing this bug report
Thank you.

closed 2024-09-16 by ButtonBasher92

Joseph-DiGiovanni, 2024-09-28:
Thank you very much. I have notified the developer of the Methylene PS2 mod of this update and he has provided me with a firmware update to test it. It seems to work perfectly!

I'm going to leave the patched firmware here since it likely won't be available on the official Methylene mod site until it undergoes further testing. Use at your own risk of course: 
[BlueRetro_playstation-v24.09_Methylene.zip](https://github.com/user-attachments/files/17174166/BlueRetro_playstation-v24.09_Methylene.zip)