darthcloud/BlueRetro #1022: Support request for Brook Wireless Fighting Board

ShinKaz, 2024-06-06:
### BlueRetro firmware version

[24.04] 2024-04-24 but stock 1.9 too

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

retroscaler

### System used

Supergun

### Bluetooth controller brand & name

Brook Wireless Fighting Board

### What is problem? (only list ONE problem per report)

Hello, I was wondering if you could add support for this board

https://www.brookaccessory.com/detail/41130595/

### What did you expect to happen?

it seems to be able to connect but no button register

### Attach files like logs or Bluetooth traces here

seems I can't upload the trace file here :( it's a .pcapng or is it not the right file ?

https://gofile.io/d/qiz5RE
please tell me a better place or email to send it

button presses are in that order :
U D L R 1P 2P 3P 4P (top row buttons of the arcade stick) 1K 2K 3K 4K (bottom row of the arcade stick) SEL STA "Home/PS/XB" touchpad key (and touchpad+left, TP+right)
[image]

darthcloud, 2025-01-24:
I made a new way to debug issue like this in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01), could you try to get me a debug trace following theses instructions?
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

Thanks

ShinKaz, 2025-01-28:
Thanks a lot for getting back to me, I hope I actually recorded it right ^^ (any chance you can make recording debug mode buttons blink on the adaptator's LEDs in a different way ? (might be useful but harder to implement?)
used PS4/Default mode. same buttons as above

[BrookWirelessFightingBoardPS4Mode.zip](https://github.com/user-attachments/files/18580657/BrookWirelessFightingBoardPS4Mode.zip)

darthcloud, 2025-02-01:
Do you have still have your original winows trace? The gofile link is dead.

Simply zip the pcapng to be able to upload here.

ShinKaz, 2025-02-01:
I think this is it but not 100% sure, sorry
[brook wireless controller mode.zip](https://github.com/user-attachments/files/18630004/brook.wireless.controller.mode.zip)

darthcloud, 2025-02-01:
Ok that look like the PS4 mode. I think our best bet is to try making the 2nd mode shown in the 2nd blueretro trace work. Is that default mode?

Can you get windows trace of that mode?
https://github.com/darthcloud/BlueRetro/wiki/Bluetooth-HCI-trace-with-Win10

I like to have a working trace of that mode to compare with your 2nd BlueRetro trace.

No need to press all the buttons, make sure to capture the first connection (pairing process) and then turn off the controller and then turn it on again to capture the reconnection.

ShinKaz, 2025-02-02:
thanks for the help.
sadly I gotta admit I don't know what I did or I am doing half the time with this board, so I son't understand what I'm doind really or what mode it is that I recorded in june. of if i recoreded anything in that trace or a few days ago. (this applied to the reconnection thing)

any chance for the PS4 mode in the recording a few days ago ?

https://www.brookaccessory.com/detail/41130595/

if I understand correctly, I need to plug the board on my PS4 and press HONE to have i set as PS4 the next time I use wireless
the other mode (?) probably was switch mode, that I used prior by mistake ? (same, gotta plug to a switch console dock, press HOME, then the board will be in switch mode next time

there's also a PS3, X-input Wired, X-input Wireless mode

only the ps4 mode seems to connect to the blue retro adapter (from trying to sync and watching leds blinking) but not buttons seem to register, not sur if I can see a bt address or mode active from bluretro web page

my understanding is that the ps4 mode has the least input lag, and this is what I need for my setup (arcade boards>HAS supergun(2 DB15 with 2 bluretros)>RetroTink 4K>LG C9 tv)

it's been 6 month but I don't think any onther mode connect to the bluretro (some do on windows with pcap obviously, since you seem to have detected another mode in my pacap from 6 month ago)

I wish Tim Huang or anyone from the Brook Team would help us, they're usually supper reponsive on social media, but got no answer on twitter in june when I asked, might have missed it... it's sad because both of your projects are the most used in the arcade field respectively.

being able to plus controller to register those via the bluretro's usb-c port would be awesome too, but probably impossible is my guess.

"How To Use
Wired use

        1. Hold down the button

        - 1P for PS3

        - 2P for PS4

        - 1K for NS

        - 2K for PC(XID Wired)

        - 3K for PC(XID Wireless)

        2. While holding down the button, connect the board to according console to use

 

Note: The default is in PS4 mode if the designated button is not hold down upon plug-in, and it can be simply connected to PC without pressing any button.

 

Wireless use

PS3/PS4/NS: Wired used for the first time. For wireless use in the future, press the PS key, the board will reconnect with the console.

PC: Hold down SHARE, then hold down PS Key for 2 seconds. If you have installed Player LED, then you’ll see the LED intermittently flash, which shows the board has entered wireless pairing mode. At the moment, you should see « Wireless Controller« in your Bluetooth device list to pair."

thanks for all your help.

but to anyone reading this, don't waste your time any money setting this up, just solder a dualsense in your stick and enjoy already working support and the least amount of lag measured today, I'm sure someone can whip up an helper flex-board to solder super easily on the right pins and plug to a brook/standard port to DB-15/NeoGeo port.

(also I'm french, if that helps in anyway, are you too ? hehe)

ShinKaz, 2025-02-03:
> Ok that look like the PS4 mode. I think our best bet is to try making the 2nd mode shown in the 2nd blueretro trace work. Is that default mode?
>
> Can you get windows trace of that mode? [...]

the trace with a 2 is from a second adapter, same stick, is it different or better? can you tell me the adapter name?

I will, try, but not sure if the right mode👍