darthcloud/BlueRetro #1027: Master System on Krikzz Mega Everdrive Using Wireless Controller

Narcotic82, 2024-06-16:
### BlueRetro firmware version

24.04

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

Internal install DIY

### Manufacturer

esp32

### System used

SEGA MegaDrive Genesis

### Bluetooth controller brand & name

8bitdo M30

### What is problem? (only list ONE problem per report)

When I start a master system game, the buttons do not work. On the other hand, if I connect the original controller, they do.

### What did you expect to happen?

That it works

ca7bc0c5f2, 2024-07-20:
I'm not sure this will address your issue how you hope, but a work around to this behavior I found is to boot the MasterSystem game, reboot the BlueRetro, change the hardware type to Parallel 2P (MasterSystem) reboot BlueRetro again, and pair your controller. I found the same situation with an external HW1 AIO adapter.

Narcotic82, 2024-07-20:
> I'm not sure this will address your issue how you hope, but a work around to this behavior I found is to boot the MasterSystem game, reboot the BlueRetro, change the hardware type to Parallel 2P (MasterSystem) reboot BlueRetro again, and pair your controller. I found the same situation with an external HW1 AIO adapter. My testing was done with a MegaSD.

Thank you very much, changing to Parallel 2P now the controller is working :) 
Is there any way to implement pressing the home button to change the controller mode?

ca7bc0c5f2, 2024-07-21:
> Thank you very much, changing to Parallel 2P now the controller is working :) Is there any way to implement pressing the home button to change the controller mode?

I'm no developer. Just another user. I second that as a feature request though. Switching could be useful in backwards compatibility scenarios. I think even the Saturn has some degree of controller backward compatibility with the previous consoles, despite the connect change. There might be some scenarios where SNES or NES users would want to do something similar since those controller types are very similar as well.

closed 2025-01-24 by darthcloud