darthcloud/BlueRetro #1031: Request to Add Suport for Gamesir G4 Pro

IgorPereira1997, 2024-06-28:
### BlueRetro firmware version

v24.04

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

darthcloud/BlueRetro Team

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Gamesir G4 Pro

### What is problem? (only list ONE problem per report)

The controller is identified correctly, but the connection between G4 Pro and BlueRetro constantly is turned off for 3-5 seconds until the reconnect again, making it hard to play with this specific controller into PS2

### What did you expect to happen?

Expected the controller to, once connected, remain connected with BlueRetro in a stable form so playing with it is possible without problems

### Attach files like logs or Bluetooth traces here

[gamesir_g4_pro_trace.zip](https://github.com/user-attachments/files/16035750/gamesir_g4_pro_trace.zip)

Order of buttons pressed:

X, Y, A, B, D-pad (left, up, right, down, in that order), Start, Select, RB, RT, LB, LT, L-Analog (left, up, right, down, in that order), R-Analog  (left, up, right, down, in that order), RSB, LSB
[image]

Page of the controller: https://www.gamesir.hk/pages/g4pro-tutorial

Image ans location of controller's buttons:

[image]

darthcloud, 2025-01-24:
I made a new way to debug issue like this in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01), could you try to get me a debug trace following theses instructions?
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

Thanks

IgorPereira1997, 2025-01-25:
I updated the firmware and used the new Debug way. Sending you the trace in zip format and the order of button configuration was this one:

Up, Down, Right, left, Y, B, A, X, R3, L3, L1, L2, R1, R2, L-Analog (Up, Down, Right, Left), R-Analog (Up, Down, Right, Left)

[br_debug_trace.zip](https://github.com/user-attachments/files/18548004/br_debug_trace.zip)

I'll test if the connection remains stable with the new update, but in any case, this is the debug trace.

darthcloud, 2025-01-29:
I don't see the disconnection in the trace. Can you make another trace, this time wait for the controller to disconnect by itself and then save the trace.

If you already did that then is possible that the trace ran out of space.

Petruaka, 2025-02-03:
@IgorPereira1997 Friend, did you manage to get it working? I have your controller and I would like to be able to use it with blueretro, also with my Thunderobot G45 Pro and Flydigi Vader 4 Pro controllers, but I still have no references on that.

IgorPereira1997, 2025-02-03:
I have been busy this week, had time just now. But here's the debub trace:

[br_debug_trace (1).zip](https://github.com/user-attachments/files/18635208/br_debug_trace.1.zip)

So, i did just like you asked. It lost the connection two times, just so you could trace it a little better. after the first time, i pressed the Left Analog Stick upwards two times, just as a reference to you. If you need anything more, just let me know

IgorPereira1997, 2025-02-03:
> [@IgorPereira1997](https://github.com/IgorPereira1997) Friend, did you manage to get it working? I have your controller and I would like to be able to use it with blueretro, also with my Thunderobot G45 Pro and Flydigi Vader 4 Pro controllers, but I still have no references on that.

Unfortunatelly no, I still have the issue, so It is basically impossible to play with this particular controller for now

darthcloud, 2025-02-05:
Can you try this beta:
[v25.04-beta-19-g5fec5fe8_hw1_universal.zip](https://github.com/user-attachments/files/18665970/v25.04-beta-19-g5fec5fe8_hw1_universal.zip)

make sure you disable debug mode before updating

If it doesn't work make me another debug trace.

Thanks!

IgorPereira1997, 2025-02-16:
Sorry about the long time wihtout return, i've been pretty busy. I tested the beta and unfourtunatelly the problem still persists. It loses the connection and "stucks itself" in the same button(s) i was pressing when the connection cutted out. I'm sending a new debug trace, now i let the cut and reconnect happen three times, while emulating playing a racing game, pressing LPAD left and R2 buttons, with different pressures.

[br_debug_trace_16_02_2025.zip](https://github.com/user-attachments/files/18816641/br_debug_trace_16_02_2025.zip)

Thank you for keep trying to solve this issue!

darthcloud, 2025-02-17:
Thank for testing!

Can you try this new one:
[v25.04-beta-20-g0c497a1e_hw1_universal.zip](https://github.com/user-attachments/files/18818771/v25.04-beta-20-g0c497a1e_hw1_universal.zip)

IgorPereira1997, 2025-02-22:
Well, there are some updates. On The PS2 (I have a slim 90000 series) it beaheved a litte different. Instead of being stuck, it sees the joystick was disconnected and after a couple of seconds it reconnects again, but it gets "stuck" on the last action i had pressed when the cut off happend. After that, i've came to generate a new debug trace, but to my surprise, it interestingly never disconnected, so i coudn't reproduce the issue. I thought of two things: The feed by controller ports might not be enough for blueretro work, or because i played using a usb hdd, it might "steal" some of the available energy. I don't know if this is indeed a factor, but found very interesting it happened. here's the debug trace:

[br_debug_trace (1).zip](https://github.com/user-attachments/files/18924977/br_debug_trace.1.zip)