darthcloud/BlueRetro #1038: Request to Add Suport for Razer Raiju Mobile

JonDysania, 2024-07-18:
### BlueRetro firmware version

v24.04

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

myself with ESP32

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Razer Raiju Mobile

### What is problem? (only list ONE problem per report)

Suport for Razer Raiju Mobile

### What did you expect to happen?

Razer controls would be a great addition, the razer raiju mobile is cheap and easy to get compared to the other razer controls. A great controller for blueretro. Page-> https://mysupport.razer.com/app/answers/detail/a_id/3590/
[image]

### Attach files like logs or Bluetooth traces here

Order of buttons pressed:
X, A, B, Y, D-pad (left, up, right, down, in that order), Start, Select, R1, R2, L1, L2, 
L-Analog (left, up, right, down, in that order) L3,     
R-Analog (left, up, right, down,in that order) R3
Adding pcapng and hexvalues:
[razer_raiju_mobile.zip](https://github.com/user-attachments/files/16286807/razer_raiju_mobile.zip)

darthcloud, 2025-04-16:
Feel free reopen once you got the trace.

JonDysania, 2025-05-26:
Debug trace made in version  25.04 hw1, Razer raiju mobile has 2 BT modes, I include both bin files.
the button pressed order is the same:
X, A, B, Y, D-pad (left, up, right, down, in that order), Start, Select, R1, R2, L1, L2,
L-Analog (left, up, right, down, in that order) L3,
R-Analog (left, up, right, down,in that order) R3
[Razer Raiju Mobile debug trace.zip](https://github.com/user-attachments/files/20443746/Razer.Raiju.Mobile.debug.trace.zip)