darthcloud/BlueRetro #1046: Request to add : Wireless Controller for N-SL Model SW001 w/ GameCube

ShdwDncR, 2024-07-24:
### BlueRetro firmware version

v24.04_hw1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

myself with ESP32 DIY

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Wireless Controller for N-SL Model SW001

### What is problem? (only list ONE problem per report)

Not pairing with GameCube DIY ESP32, I have no problems pairing other Controllers like PS4 and Google Stadia.

### What did you expect to happen?

Pair, tried different firmware's HW1, no luck.  

### Attach files like logs or Bluetooth traces here

[Wireless_Controller_for_N-SL-Model-SW001.zip](https://github.com/user-attachments/files/16366991/Wireless_Controller_for_N-SL-Model-SW001.zip)

ShdwDncR, 2024-07-24:
https://m.media-amazon.com/images/I/D1yT0ybiyDL.pdf

ShdwDncR, 2024-07-25:
I forgot to add the keys pressed when logged the BlueTooth Traces

Y, X, B, A, -, +, T, 0,
D-Up, D-Down, D-Left, D-Right
RStick-Up, D, L, R
LStick-Up, D, L, R
LTrigger, RT,
BottomTriggerLeft, BTR

ShdwDncR, 2024-07-28:
This guy dose not give a about your request. His hungry like a JOO! So how much you want for my controller to work ?

darthcloud, 2025-04-16:
Feel free reopen once you got the trace.

closed 2025-04-16 by darthcloud