darthcloud/BlueRetro #1054: PSX game rumble weak and strong nearly same

mhmetalheart, 2024-08-06:
### BlueRetro firmware version

v24.04_hw1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

DIY

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

sony PS3 PS4 & 8 bitdo SN30 pro

### What is problem? (only list ONE problem per report)

Game: MGS1 (japan ver. But actually, I don't think that's the game version issue.). In option, can test the rumble. If use the original PSX controller, can clearly feel the weak & strong different.
But on BR with PS3, PS4 or SN30 pro, nearly same. I move left right (change weak <> strong) very fast, I can feel they're nearly no any difference. 
Also, on the Ape Escape (also japan ver. named: saru get you!). If slip by the banana skin, can feel a very weak rumble by original PSX controller. But useing BR, the rumble nearly same with 'catch you'. 

### What did you expect to happen?

make the rumble have clearly different.

darthcloud, 2024-10-06:
Can you retest with this firmware?

I completely overhaul the rumble support in v24.10:
https://github.com/darthcloud/BlueRetro/releases/tag/v24.10

mhmetalheart, 2024-10-12:
Just tried again. No, this time no rumble. Both MGS and APE. 
I tested: 8BITDO SN30, PS3, PS4. All of them, no respond. I can make sure I press the button to analog model. But then, nothing happened.

darthcloud, 2024-10-12:
Did you enable rumble in the BlueRetro config?
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#26---example-enable-rumble

mhmetalheart, 2024-10-12:
damn! how foolish I'm. I don't know why but the rumble option was off.
Yes, now the weak and strong was totally different on MGS. I got a very weak rumble on APE when slip by the banana skin. Good work!
I will report another issue, but maybe that's our wrong using. Hope can give me some idea.

closed 2024-10-12 by mhmetalheart