darthcloud/BlueRetro #1063: Bitfunx blueretro nor working on OPL PS2

Nick5779, 2024-08-25:
### BlueRetro firmware version

I don't know

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Bitfunx

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Sony dualshock 4

### What is problem? (only list ONE problem per report)

My bitfunx's blueretro works normally with disc based games, but when I launch a game via OPL it says the controller is detached

### What did you expect to happen?

Launch a game thought OPL, use my dualshock 4 and not having any problems lol

patrikmckane, 2024-08-27:
I own a bitfunx blueretro updated to 24.04 firmware. no issues there. its congigured for playstation and i use dualschock 3

Nick5779, 2024-08-27:
Do you know if a non original dualshock 3 will work???

patrikmckane, 2024-08-27:
I dont know honestly. I dont háve any fake ps3 controller and when they Come I get rid of them as soon as possible.

x1nf3ctAdusX, 2024-08-29:
I updated my blueretro, which is in fact form retroscaler, it's the "mini" version, and my dualshock 4 is still listed as Controller removed in the game I'm trying to play that's Midnight Club 3 DUB EDITION REMIX lol. I just want to play my game with my dualshock 4 at least or even my "fake", non original dualshock 3.

darthcloud, 2025-05-13:
For me it work but I'm using their old official release from 2021:
sha1sum OPNPS2LD-v1.1.0.ELF
be594ff83a81486c20b3f7276bcef3041bea29f3 OPNPS2LD-v1.1.0.ELF

I can't get the latest betas of OPL to work on my FHDB setup, they just give me a black screen.

One thing you can try is disabling PADEMU feature at both the Global and game level in OPL settings.

darthcloud, 2025-07-13:
Feel free reopen if this didn't help

closed 2025-07-13 by darthcloud