darthcloud/BlueRetro #1073: Stadium Controllers - so not switch off or stay paired

haydnkilyan, 2024-09-09:
### BlueRetro firmware version

24.04

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

Internal install modkit

### Manufacturer

Laserbear

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Google Stadia

### What is problem? (only list ONE problem per report)

Controller pairs fine 
When I switch of the console, controller stays on until I manually turn it off
When I switch the console and controller on again - they are no longer pair
They need to be re-paired everytime

### What did you expect to happen?

I hoped that when I turned the console off, the controllers would auto turn off (my 8BitDo Pro 2s do this)
I hoped that when I turned the console/controllers on they would remain paired (not need to be re paired every time)

kevums, 2024-09-16:
Ditto on issue. Occurs across two controllers and two blue retro modules. Additional issues exist within the same setup with regards to Stadia controllers staying connected; will submit seperate req.

darthcloud, 2024-10-06:
For the controller not turning off, its a bug on the Stadia FW. Nothing I can do. They ignore the turn off command.

The other issue is a dup of #622

closed 2024-10-06 by darthcloud