darthcloud/BlueRetro #1080: Wario World GameCube - Right Trigger

haydnkilyan, 2024-09-23:
### BlueRetro firmware version

24.04

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install modkit

### Manufacturer

Laserbear

### System used

Nintendo GameCube

### Bluetooth controller brand & name

8Bitdo Pro 2 / PS Dual shock 4

### What is problem? (only list ONE problem per report)

In Wario World (only) the right trigger does not work. I have seen others report this issue. I have tried the merged trigger preset. This works well on all other games I have tried (for example Mario Sunshine) but Wario does not respond to this input at all

### What did you expect to happen?

Right trigger is unresponsive

Wild-Dog, 2024-10-01:
Same exact issue on ps3 controller.

darthcloud, 2024-10-06:
duplicate #1015

closed 2024-10-06 by darthcloud