darthcloud/BlueRetro #1082: Please support 8BitDo Mod Kit for Original NGC Controller

rayrayrayraydog, 2024-09-27:
### BlueRetro firmware version

24.04

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install modkit

### Manufacturer

GameCube Blue Retro Internal Adapter (pre-V3)

### System used

Nintendo GameCube

### Bluetooth controller brand & name

8BitDo Mod Kit for Original NGC Controller

### What is problem? (only list ONE problem per report)

I have been testing the 8bitdo GC Mod Kit with the LaserBear Internal GC adapter.  https://shop.8bitdo.com/products/mod-kit-for-original-ngc

On firmware 24.04: Pairing the GC Mod Kit in Switch mode mostly just works (button mapping wrong / no rumble), but a limitation of this mode is no analog values for the L/R triggers.  Pairing the GC Mod Kit in Android mode leads to not buttons being registered at all in the system.

On firmware v1.8-rc4 hw2 gamecube (shipped w/ device): Buttons are working fine including analog triggers, with some limitations (Digital L/R buttons cannot be mapped, rumble not working).

### What did you expect to happen?

I expect button presses to work after pairing the controller in its Android mode (B+A pairing) on the latest firmware.  This is the only mode that reports analog values for the L/R triggers.

EeveeEuphoria, 2024-09-28:
I'm also experiencing these same issues, albeit with the RetroTime GC BlueRetro Wireless Receiver. The analog stick calibrations are also incorrect, along with incorrect buttons (X and B are swapped, Z can not be binded at all). Direct support for this would be nice, as I bought this before knowing 8bitdo would make this.

jamieshaw, 2024-10-01:
Have just received my NGC Mod Kit and after installation, was disappointed to know this doesn't work in "Android Mode" (but assume it's nothing more than software for the mod kit or LaserBear board).

Looking forward to any updates/settings that can accommodate this 🙂

theretrobristolian, 2024-10-02:
Hey, Just a quick message to say I'm experiencing exactly the same thing, if it pairs in 'Switch' mode, the buttons are mapped incorrectly, no rumble, triggers don't seem to be analog, if in android mode I think it pairs but absolutely no response on the console (gamecube) at all with the laser bear BlueRetro mod.

I'm been subscribed via GitHub to sponsor the project for a while and thing BlueRetro is an outstanding suite that can cover so much, this would be amazing to see this issue find a resolution.

more than happy to assist, test in anyway as I've bought 4 of the 8bitdo kits and currently can't use them. In the mean time I have now ordered the 8bitdo receiver as well so I can compare results.

Thanks!

rayrayrayraydog, 2024-10-02:
For anyone helping test, you can use this DOL app via Swiss to test button mappings: https://github.com/corenting/GC-Controller-Test

hosser, 2024-10-02:
+1

I'm using 24.04 fw with the 8BitDo GCN modkit and the Retro Time receiver on a GameCube console. The B and X buttons were switched (which can be easily fixed in blueretro config) but the L/R triggers act as Z, and the Z button itself doesn't work at all. I haven't been able to fix the triggers so for now I am sticking with my GBros adapter. Would love to be able to use this in future with a firmware update.

EeveeEuphoria, 2024-10-06:
Giving the newest commit [5f9608f](https://github.com/darthcloud/BlueRetro/commit/5f9608f807ac55af455f5f8056950ce7ac714327) a whirl in regards to all of this (yes I've been obsessively refreshing the page just to see when this would get fixed), and it does seem to mostly work! However, vibration doesn't work in android mode, which is unfortunate. I have the GC 8bitdo adapter arriving tomorrow, and I'll see if vibration works on that (as far as I can tell, nobody online has been able to verify if that even works in android mode). Another thing is that it seems that sometimes, the left stick calibration is wrong, and I need to unplug the adapter and plug it back in to get it to work.

darthcloud, 2024-10-06:
Don't use FW I haven't released myself. It's not for a good reason. If you brick your system using unreleased software no one will help you.

I will post here when it's ready.

EeveeEuphoria, 2024-10-06:
I got my 8bitdo adapter in, and it does support rumble (albeit weakly for some reason). Additionally, the ranges it gives go from 0 to 128, as opposed to normal gamecube controllers (and the blueretro) reporting 0 to 100.

So, there's that.

darthcloud, 2024-10-06:
BTW the rumble work for me on the beta build, maybe you dont have rumble configure properly in the blueretro setting.
(It's very weak compare to wired gc one. so you need a somewhat strong effect to test it)
On Blueretro the range on the controller will be the proper one.

You need to use the default gamepad only or default gamepad/keyboard for the end trigger button to be used. (If you got the merge preset, the end is simulated base on analog trigger position rather than end button)

EeveeEuphoria, 2024-10-06:
The trigger buttons worked fine for me and detected like normal, no anomalies I could find there on the blueretro. I also followed your advice, I didn't know rumble had to be enabled seperately, and it works! I will say, it is actually somehow a little bit stronger than the 8bitdo adapter. Not by much, but it's noticeable enough to me.

darthcloud, 2024-10-06:
I'm done with testing, so this will be the final release for v24.10 later today:
https://github.com/darthcloud/BlueRetro/actions/runs/11203416517

darthcloud, 2024-10-06:
Pairing guide:
https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#16---8bitdo-ngc-modkit-paring-guide

darthcloud, 2024-10-06:
See official release: https://github.com/darthcloud/BlueRetro/releases/tag/v24.10

closed 2024-10-06 by darthcloud

rayrayrayraydog, 2024-10-06:
> See official release: https://github.com/darthcloud/BlueRetro/releases/tag/v24.10

Woohoo!  Will check it out today!

mjenkins-fcb, 2024-10-06:
This is fantastic work thank you! Just updated my Laserbear Internal Blue Retro and everything works apart from L and R button presses. So analog works but not digital.

rayrayrayraydog, 2024-10-06:
Just fired this up and tried it.  Everything is working for me, including rumble and the L/R analog triggers/buttons.  The only issue was B and X were swapped by default, easy enough to configure that of course.

darthcloud, 2024-10-06:
not supposed to be swapped if you use default presets, you probably had left over prior configuration.

rayrayrayraydog, 2024-10-06:
> not supposed to be swapped if you use default presets, you probably had left over prior configuration.

Oh that makes sense.  Thanks.  I'll restore defaults and pair again.

rayrayrayraydog, 2024-10-06:
Confirmed.  Everything works perfectly immediately when connecting after setting defaults.  Thanks!!

mjenkins-fcb, 2024-10-07:
> This is fantastic work thank you! Just updated my Laserbear Internal Blue Retro and everything works apart from L and R button presses. So analog works but not digital.

I installed the L and R button pad things upside down! All works perfectly now.

theretrobristolian, 2024-10-07:
Just to say here formally a big thank you for the quick turn around, I've updated and successfully paired - now to try some games :-)

Skatendo16, 2024-10-09:
For the 8bitdo NGC controller mod, I followed the instructions and it seems to be working in android mode, but rumble isn't working. I use the v24.10_hw1_gamecube.bin firmware update. Was that the correct one?

Papy-G, 2025-11-16:
> I got my 8bitdo adapter in, and it does support rumble (albeit weakly for some reason). Additionally, the analog stick ranges it gives go from 0 to 128, as opposed to normal gamecube controllers (and the blueretro) reporting 0 to 100. Also of note is the fact that, for some reason, the 8bitdo adapter starts off with an offset of -01, -01 for the left stick, and 00, -01 for the c-stick, while blueretro doesn't suffer from this if you don't touch the analog sticks as the first input.

The available voltage for rumble is lower when working off a battery, putting a phone's rumble motor may be a way to better feel the vibration, while it may not feel like the original GC's controller then.
The higher value you get at the borders of the range, the shorter the "deadzone" is on sticks, so you can see those values. I suspect you can hardly walk in WindWaker with such a setting.

> You need to use the _default gamepad only_ or _default gamepad/keyboard_ presets for the end trigger button to be used. (If you got the merge preset, the end is simulated base on analog trigger position rather than end button)

It's not preset this way for me.
Can't we just get the real end buttons mapped 1 to 1 instead of simulating it off the analog trigger? Is it possible to work hand in hand with 8BitDo so we can just get plug-and play, proper 1 to 1 gamecube controller to gamecube support as default?