darthcloud/BlueRetro #1087: Getting Blueretro to work on ESP32-­PICO-­V3

olderzeus, 2024-10-04:
### BlueRetro firmware version

v24.04 hw1 parallel_1p

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

DIY

### System used

Handheld

### Bluetooth controller brand & name

ESP32­-PICO-­V3

### What is problem? (only list ONE problem per report)

I want to make a built-in mini ESP32 module for my DS Lite, using ESP32-PICO-V3, Parallel 1P mode directly connected to the button contacts. But it can't start after flashing the firmware, including custom PICO-V3 firmware provided by #156

### What did you expect to happen?

Getting Blueretro work on ESP32-­PICO-­V3

### Attach files like logs or Bluetooth traces here

[pico_v3.txt](https://github.com/user-attachments/files/17254354/pico_v3.txt)
[v24.04.txt](https://github.com/user-attachments/files/17254356/v24.04.txt)
[v24.06.txt](https://github.com/user-attachments/files/17254358/v24.06.txt)

closed 2024-10-04 by olderzeus

konwektor, 2024-10-15:
From Your logs "I BOD: Brownout detector was triggered"
You have problems with power supply - voltage is dropping too much and  brownout is triggered - resets chip.

olderzeus, 2024-10-15:
> From Your logs "I BOD: Brownout detector was triggered" You have problems with power supply - voltage is dropping too much and brownout is triggered - resets chip.
Yes, I have solved this problem and adjusted the relevant content to make it work properly. ESP32-PICO-V3
will be recognized as EFUSE_RD_CHIP_VER_PKG_ESP32PICOD4(CPU V3.1)