darthcloud/BlueRetro #1088: Wavebird controllers sometimes don't work depending on order plugged in?

bkacjios, 2024-10-07:
### BlueRetro firmware version

24.10

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install modkit

### Manufacturer

Laserbear

### System used

Nintendo GameCube

### Bluetooth controller brand & name

N/A

### What is problem? (only list ONE problem per report)

This is a weird one, and I'm not even really sure if this is a software bug or a hardware bug with Laserbear's Blue Retro Internal Adapter V3.

The best way to explain it, plugging in multiple wavebird receivers sometimes causes controllers to not be detected by the gamecube. This only seems to happen if the receivers are in ports that are next to each other, for example 1 & 2, 2 & 3, or 3 & 4. 1 & 4 also seem to cause the same issue. An odd thing to note is that plugging in 3 receivers in any order seems to fix it. 4 all plugged in also works as well.

I tested this by putting all my receivers and a single wavebird on channel 1.

https://www.youtube.com/watch?v=oMZ3yIJpBnk

### What did you expect to happen?

All receivers to work no matter what position or how many are plugged in.

darthcloud, 2025-04-19:
This is definetly an HW issues best to contact Greg, maybe he can test to see it's something specific to your board or not.

closed 2025-04-19 by darthcloud

bkacjios, 2025-04-19:
Ah heck. Looks like I have actually have a v2 of the internal mod, and they now released v3 with better port detection.. I'm guessing it fixes this.

Thanks for looking though.