darthcloud/BlueRetro #1093: PSX Porsche Challenge rumble not working on v24.10 other PSX titles ok

patters-match, 2024-10-11:
### BlueRetro firmware version

24.10

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

PSUnoRetro

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Sony DualShock 4

### What is problem? (only list ONE problem per report)

I reported #155 long ago and you recently made changes to support dual motor rumble. I updated and tried it but strangely the rumble in PSX Porsche Challenge is not working, at least on PS2 via backwards compatibility (I don't have a real PSX to try). This works fine with a regular DualShock2, so if the blueretro emulates this then it ought to work. I tried setting the system type from Auto to PS2 in the advanced config but that didn't help. Rumble support in Gran Turismo works fine for instance, so it may be some kind of edge case.

### What did you expect to happen?

Rumble in game when the Analog mode is toggled for the controller.

darthcloud, 2024-10-11:
What region of that game? I tried US one on a PS1 and no rumble on either OG analog, DS1 or DS2

darthcloud, 2024-10-12:
Ok saw its was PAL in other bug, tried that and rumble does work. I'll check why BlueRetro doesn't work.

darthcloud, 2024-10-12:
Work now with this build:
https://github.com/darthcloud/BlueRetro/actions/runs/11302189293

patters-match, 2024-10-12:
Thanks, will try today. I should also have mentioned that the analog control worked fine.

https://github.com/darthcloud/BlueRetro/actions/runs/11302189293

patters-match, 2024-10-12:
Woohoo - success! 😃  Thanks for the rapid response!

closed 2024-10-12 by darthcloud