darthcloud/BlueRetro #1100: Blueretro does not work without a power cable

Yura993, 2024-10-17:
### BlueRetro firmware version

24.10

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Aliexpress 

### System used

Sony PlayStation 2

### What is problem? (only list ONE problem per report)

After updating the firmware to 24.10, blueretro adapter does not work without a power cable on ps2 slim, everything worked on the old firmware

### What did you expect to happen?

.

lux778999, 2024-10-19:
not to be that guy, but i think this should be adressed to retroscaler github instead of darthcloud's, and idk if you know this already, i assume you do, but the retroscaler products have their own updates at the retroscaler github, i have a blueretro mini and this one doesnt work if the update that i put into it isn't the retroscaler one

Yura993, 2024-10-19:
> not to be that guy, but i think this should be adressed to retroscaler github instead of darthcloud's, and idk if you know this already, i assume you do, but the retroscaler products have their own [...]

Thanks for the answer, but the retroscaler firmware does not solve the problem, the adapter works well on darthcloud's firmware, the only problem is that when connecting the adapter to the ps2, it does not turn on without connecting an additional USB power cable, although the adapter previously worked well without it, I will be very grateful if I know how to solve this problem

basbase, 2024-11-21:
Same issue. Don't know which firmware since it is straight out of the box. It does work in a fat ps2 without a usb cable.

darthcloud, 2025-04-22:
This isn't something that can be fix by software.

It's one of those possibility:
#1: Defecttive adapter
#2: PS2 rumble fuse is blown
#3: RetroScaler design draw too much power on some slim system.